JoelBender/bacpypes #163: Unable to write a SequenceOf property

calligp, 2018-02-02:
I am implementing write operation on notification-class recipientList to comply with BTL implementation guidelines. In fact, the requirement is to write the full property data with `WriteProperty` operation but this property is a `SequenceOf` and actually the support of write on this property type seems not be supported.

The issue is pretty similar than the writing of an `ArrayOf` from issue #147.

I have apply a similar patch to write `SequenceOf` like it is already done for `ArrayOf`. See attached file.

[0015_WriteSequenceOf.txt](https://github.com/JoelBender/bacpypes/files/1689579/0015_WriteSequenceOf.txt)

Nevertheless I have a question regarding my patch content:
In `constructeddata.py` source file, the `SequenceOf` is not a subclass of `Sequence`. I think it is an error but I want your opinion on?

JoelBender, 2018-02-03:
There once was an **Annex C - Formal Description of Object Type Structures (Informative)** that described the properties of objects using ASN.1:
```
CALENDAR ::= SEQUENCE {
    ...
    date-list   [23] SEQUENCE OF BACnetCalendarEntry,
    ...
    }
```
And BACpypes turned that into `SequenceOf(CalendarEntry)` which could also be used for the ASN.1 for the APDU's.  There were also definitions like this:
```
    action [2] SEQUENCE OF BACnetActionList, -- accessed as a BACnetARRAY
```
Which I turned into `ArrayOf(ActionList)`.  These days the Annex is gone and the object definition now looks like this:
```
    BACnetLIST of BACnetCalendarEntry
```
And that should be turned into `ListOf(CalendarEntry)`.

So your basic instinct is correct to make these properties be constructed types like arrays, but `Sequence` is more like a structure, so `SequenceOf` is not a subclass of `Sequence`.  However, if there was a new `List` type, then `ListOf(cls)` would be a derived type, and that concept would be distributed around the rest of the code base.

This is a significant API change.

JoelBender, 2018-02-04:
First, I agree that checking to see if a property is a SequenceOf(something) is harder than it should be, and second I'm rethinking the notion that it's a significant change.  You could check `self.datatype in bacpypes.constructeddata._sequence_of_classes` (thanks to the dictionary being private to the module, you can't just import it).

Digging into your patch a little more, the `if arrayIndex is not None` should immediately fail with `propertyIsNotAnArray`, or simply skip that check because it will fail later.  If the value is _not_ a list, then that is some other kind of error (and I need to look up what error that is).

JoelBender, 2018-02-13:
I had a hack, but it feels wrong.  I'm back to adding a new `List` class and `ListOf` and change all the occurrences of `SequenceOf` to `ListOf` for the object definitions, but leave the `SequenceOf` definitions for the APDUs.  But that also means I'm also going back to this being a 0.17.x change rather than stuffing anything more in the 0.16.7 release.

calligp, 2018-02-13:
You right, my patch is too much like "copy-paste" of `ArrayOf` implementation. And my second error is regarding the management of `arrayIndex` parameter. Indeed for sequence only the complete list of element can be updated on not only one element.

I tested you branch `163-write-a-list` and it seems that it works perfectly with my server implementation (after replace all `SequenceOf` by `ListOf` type).

I saw just one minor issue in `service/cov.py` file, the `actioveCovSubscriptions`is declared with type `SequenceOf(COVSubscription)` but in `object.py`, it has type `ListOf(COVSubscription)`.
Thank you

JoelBender, 2018-02-14:
I fixed that piece, good catch.  I also added some error checking to the `Property.__init__()` method to make sure the datatype is one of the correct types just to make sure that if there's any old code out there that uses `SequenceOf` rather than `ListOf` that it "fails fast," otherwise the problem won't show up until the application is tested using the BTL tools which is really late in the development process.

JoelBender, 2018-04-07:
This is in the stage branch and ready for release.

closed 2018-04-07 by JoelBender