JoelBender/bacpypes #164: Management of deviceInfoCache in BIPForeignApplication

calligp, 2018-02-02:
I begin to use the `BIPForeignApplication` class. But I see a major difference between `BIPSimpleApplication` and the
`BIPForeignApplication` regarding `DeviceInfoCache`. Indeed both application are child of
`ApplicationIOController` but only simple application provides `DeviceInfoCache`. I think it is a oversight because
`DeviceInfoCache` is used (among other) to manage segmented APDU.

See attached patch to add `localAddress` and `deviceInfoCache` parameters to `ApplicationIoController` initialization.
[0017_ForeignApp_AddDeviceCache.txt](https://github.com/JoelBender/bacpypes/files/1689650/0017_ForeignApp_AddDeviceCache.txt)

JoelBender, 2018-02-03:
You are correct, this will go into the next release, thank you.

closed 2018-02-03 by JoelBender

calligp, 2018-02-05:
I think a trouble is present in your patch.
(Same than in issue #157) 

BIPForeignApplication can not be started because of parameter order. Indeed In my own patch I set `localAddress` parameter between `localDevice` and eviceInfoCache` in `ApplicationIoController` instanciation.

JoelBender, 2018-02-12:
This has been merged in, sorry about the extra fix to the patch :frowning_face:

commit 5a796fcc (2018-02-03): closes #164