JoelBender/bacpypes #165: Micropython support

tarakanov, 2018-02-05:
I am curious about micropython support. In case I want to develop embedded device with BACnet support. Is it hard to port bacpypes to Micropython?

JoelBender, 2018-02-05:
Based on a quick survey of the MicroPython language docs, then only thing that gives me pause is the [method resolution order](http://docs.micropython.org/en/latest/pyboard/genrst/core_language.html#method-resolution-order-mro-is-not-compliant-with-cpython) support, as long as you don't build "diamond" inheritance chains for enumerations and objects.  With enumerations it's not a big deal, but with objects it could be trouble.  I wouldn't be surprised if there aren't some other hidden dragons.

It would be nice if I could tell you to just run the tests and if they all pass you're good to go, but the test suite isn't good enough for that.  I'll poke around some more, it looks like the repository has instructions for running it locally.

kmonson, 2018-02-05:
What about the supported standard libraries? I see an import of asyncore in core.py and I don't see it in the list of modules here: http://docs.micropython.org/en/latest/pyboard/library/index.html

JoelBender, 2018-02-07:
There are lots of libraries listed [here](https://github.com/micropython/micropython-lib), and the uasyncio family of libraries looks promising.  I haven't stumbled across the procedures for including these libraries on a pyboard yet.

If you could put together the instructions for an example project that echoes a packet on the BACnet UDP socket (47808) and has a way to "schedule" something to happen with a delta time (like generating a retry for a request if the ack doesn't come back) and future absolute time (like what the schedule object does) then we can start from there.

The lowest level pieces of BACpypes are the `bacpypes.core`, `bacpypes.task`, and `bacpypes.udp` modules, everything else is a stack of objects.  Because BACpypes started in the Python2.5 days it doesn't have any of the new fangled async/await usage.

tarakanov, 2018-02-07:
@JoelBender thanks for the direction. I will try and post the result in this thread

Schade77, 2019-10-17:
Hello,

@tarakanov did you advance on the question?

I am also interesting of the possibility to have a bacnet device on micropython.

Regards,

JoelBender, 2019-10-17:
I just tried a build and it failed, so there's more investigating to be done:
```
CC build/frozen_content.c
build/frozen_content.c:14:12: error: unknown type name ‘most’
 Traceback (most recent call last):
            ^
```
The file looks like generated code and the generation process failed:
```
//
// Content for MICROPY_MODULE_FROZEN_MPY
//
Traceback (most recent call last):
  File "../../tools/mpy-tool.py", line 833, in <module>
    main()
  File "../../tools/mpy-tool.py", line 821, in main
    raw_codes = [read_mpy(file) for file in args.files]
  File "../../tools/mpy-tool.py", line 821, in <listcomp>
    raw_codes = [read_mpy(file) for file in args.files]
  File "../../tools/mpy-tool.py", line 675, in read_mpy
    raise Exception('incompatible .mpy version')
Exception: incompatible .mpy version
```