JoelBender/bacpypes #167: Support unicast for foreign device application

calligp, 2018-02-09:
# xxx Support unicast for foreign device application
Implementing foreign device behavior in my BACnet application, I notice an major difference between explanation and
bacpypes implementation. In following page ([[http://www.bacnet.org/Tutorial/HMN-Overview/sld036.htm]]), the foreign
application can "talk with any BACnet device directly without registration". Only the broadcast exchange are blocked
until `RegisterForeignDevice` service is acknowledged by BBMD device.

My tests regarding this special case are failing with following debug logs:

```
2018-02-07 11:41:20,315 - DEBUG - bvll - decode - decode <PDU 10.33.74.62 -> 10.33.74.104 :
81.0a.00.0c.01.20.ff.ff.00.ff.10.08>
2018-02-07 11:41:20,318 - DEBUG - bvll - __init__ - __init__ () {}
2018-02-07 11:41:20,318 - DEBUG - bvll - __init__ - __init__ () {}
2018-02-07 11:41:20,319 - DEBUG - pdu - __init__ - __init__ () {}
2018-02-07 11:41:20,319 - DEBUG - pdu - __init__ -     - my_kwargs: {}
2018-02-07 11:41:20,319 - DEBUG - pdu - __init__ -     - other_kwargs: {}
2018-02-07 11:41:20,320 - DEBUG - comm - __init__ - __init__ () {}
2018-02-07 11:41:20,320 - DEBUG - comm - __init__ -     - my_kwargs: {}
2018-02-07 11:41:20,320 - DEBUG - comm - __init__ -     - other_kwargs: {}
2018-02-07 11:41:20,321 - DEBUG - comm - __init__ - __init__ None () {}
2018-02-07 11:41:20,321 - DEBUG - comm - response - response (<bacpypes.bvll.OriginalUnicastNPDU object at
0x7fdc3c46f890>,) {}
2018-02-07 11:41:20,321 - DEBUG - bvllservice - confirmation - confirmation <bacpypes.bvll.OriginalUnicastNPDU object
at 0x7fdc3c46f890>
2018-02-07 11:41:20,322 - DEBUG - bvllservice - confirmation -     - packet dropped, unregistered
```

After investigation I found the potential root cause in the `BIPForeign` class from `bvllservice.py` file. Indeed
`registrationStatus` is checked before analyse if request is unicast or broadcast. This check is done in
`confirmation` and `indication` methods.

I suggest to check `registrationStatus` only for non unicast request.
See patch attached.

[0019_UnicastToForeign.txt](https://github.com/JoelBender/bacpypes/files/1711624/0019_UnicastToForeign.txt)

JoelBender, 2018-02-10:
From the standard...

> *J.5.2 BBMD Operation - Foreign Devices*
> 
> In order for a foreign device to fully participate in the activities of a B/IP network, the device must register itself with a BBMD serving one of the IP subnets comprising that network. "Full participation" implies the ability to send and receive both directed and broadcast messages.

I interpret this to mean that both unicast and broadcast messages should fail when it is not registered.  In theory this also means that the stack should probably block all incoming traffic until a registration goes out and an positive acknowledgement is returned.

Abdoubng, 2018-02-10:
> J.5.2 : In order for a foreign device to fully participate in the activities of a B/IP network

 
and the sentance taken from the bacnet.org tutorial : 
http://www.bacnet.org/Tutorial/HMN-Overview/sld036.htm

> The foreign device can, of course, talk with any BACnet device directly without registration but will only receive broadcasts if the registration procedure is followed.

Don't contradict each other, you can say that the Foreign device is participating with unicast messages but according to : 

> "Full participation" implies the ability to send and receive both directed and broadcast messages 

in order to fully participate broadcast messages are needed, which leads to a registration with a BBMD. and by definition a BBMD is just here to manage the Broadcast since routers generally forbid it.

This is how i interpret it, what do you think about it ?

JoelBender, 2018-02-12:
Well, your interpretation seems to be supported by another statement I put in the [BACnet/IP slides](http://www.bacnet.org/Tutorial/BACnetIP/sld012.html):
> Assuming that two devices know each other's IP addresses and the UDP ports they are using, i.e., their respective B/IP addresses, there's nothing that restricts them from communicating directly).

Originally I expected foreign devices to not be substantially persistent, for example they would start up and do what they needed to do and go away, and if they had all the information they needed like device address binding and routing there would be no need for broadcast services.

I went through 135.1 looking for comments that might restrict foreign devices before they registered and didn't find any, and when looking at tests like this one:
> *14.1.8 Original-Broadcast-NPDU*
> Purpose: To verify that an IUT, not configured as a BBMD, will process an Original-Broadcast-NPDU message.

It doesn't say anything different about being an unregistered foreign device.  I started a new branch with your patch, I don't think this is different enough behavior to wait for a larger (0.17) release.

JoelBender, 2018-02-17:
This works as you intended, thank you.

closed 2018-02-17 by JoelBender