JoelBender/bacpypes #168: Management of segmentation in response

calligp, 2018-02-16:
During my tests around response segmentation, I saw error in response format:
1. No one of response or request were segmented.
   This first issue is in my server implementation. I did not have instantiated an `deviceInfoCache` and provided it to `BIP*Application` initialization. So when bacpypes library was trying to send the response, no segmentation data was available.

   Do you confirm this object is mandatory for segmentation processing? In my first understanding, this object is optionnal because most of important segmentation data are provided in request header.

2. Segmentation informations provided in request header are not taken into account.
   The `DeviceInfoCache` is updated with each `iAm` message received to register segmentation information. But I saw an issue when the segmentation requirement from `iAm` message and from request header are not the same. The `iAm` content is always used and request header seems ignored.

   After some research I found in `ServerSSM` implementation two debug logs:
`- update maximum segments accepted?`
`- update maximum max APDU length accepted?`

   When I complete this function with update of `deviceInfoCache` using request header data, the segmentation is done as expected. I'm not totally sure it is the good solution but in current implementation the request header is no more available when the response is segmented.

3. When number of segment required is bigger than max accepted, the segments were sent.
   The last issue appeared when the response size is bigger than the allowed size regarding apdu length and max segments accepted. For example with a response of 150 octets, if client has maxApduLength = 50 and maxSegmentAccepted = 2. The expected result is `ABORT` message with reason `bufferOverflow`.

   But when I try it, 3 segments are received.
   After investigation, I finally found that this case is not checked in `confirmation` function of `SSMServer` class.
   I just added a check if number of required segment is bigger than client max accepted.

4. BTL specific implementation
   This is not an error in library implementation but it is a requirement from BTL implementation guideline (http://c.ymcdn.com/sites/www.bacnetlabs.org/resource/resmgr/Files/BTL_Implementation_Guideline.pdf) regarding segmentation parameter management. When the max segments accepted parameter is not provided, BTL requires to replace it by 2.

   I suggest you to add it in library implementation because it doesn't contradict the standard.

You can find my patch in attachment for more details.
[0018_Segmentation.txt](https://github.com/JoelBender/bacpypes/files/1731831/0018_Segmentation.txt)

JoelBender, 2018-04-06:
This issue screams out for test coverage like a terrified child at the bottom of a dark well.  Thank you for the patch, this has not been forgotten.

ChristianTremblay, 2018-04-06:
Bruce Wayne Unit Test.... Time to go back to the Bat Cave and code :)
https://www.youtube.com/watch?v=-ig3nsEgqNE

JoelBender, 2018-04-09:
Here's an interesting tidbit:

> 3.10 The value of Max_APDU_Length_Accepted might be different on different port

There is currently no mechanism to store the maximum size PDU that can be sent on a port, this would be a new attribute of the `NetworkAdapter`.  Looks like that if different values for the `maxApduLengthAccepted` property depend on where the request was coming from there's going to have to be a reference to the adapter stored in upstream PDUs (_yea, whatever_) and maintained all the way through the state machine code (_sheesh_) to the application layer to some `ReadProperty` function of a new special class (_yuck_).