JoelBender/bacpypes #172: Unregistered Foreign Device Application could act as a BIPSimpleApplication

ChristianTremblay, 2018-03-02:
This would allow one device to be able to act on a LAN as a simple application. And if it is needed, register to a BBMD.

Merging features of BIPForeign and BIPSimple would allow that. 

This would be an alternative to making the choice when instanciating the device (where one could choose between the classes depending on the needs at the time of instanciation)

JoelBender, 2018-03-02:
To kick things off I put a "switch" class in the `172-simple-foreign-switch` branch sandbox.  The basic idea is that you would build an instance of each of the `BIPSimple` and `BIPForeign` classes, then wrap the two of them together into a "switch" and each of the instances are "terminals."  You start out with the first instance in the stack, then you can call `some_switch.switch_terminal(n)` the other instance is in the stack.  You could also provide a BBMD instance and switch to that.

Knowing when to switch would be up to your application, as well as kicking off the foreign device registration process when that became active.  We can add additional "activated" and "deactivated" methods if you want the terminal to know when it's becoming active, or add methods for the terminal to tell the switch to make it the active one, and maybe there shouldn't be an active terminal when it's created.

I'm not convinced that "switch" and "terminal" are very good names.

JoelBender, 2018-06-01:
I changed terminals to be referenced by name rather than position, added `activate()` and `deactivate()` function lookups, and you can add and delete terminals on the fly.  Terminals no longer have to be both clients and servers, so you can have a switch at the top or bottom of a stack and not just in the middle.

It seemed like the thing to do!

JoelBender, 2018-06-20:
The new class can be moved into `bacpypes.comm` module if it does everything it should.

JoelBender, 2018-07-04:
Moved into `bacpypes.comm` and sandbox code moved into samples.

closed 2018-07-04 by JoelBender