JoelBender/bacpypes #177: BBMD request for ForeignApplication server

calligp, 2018-03-22:
In previous issue (#166), I provided to you a patch to allow foreign device to reply to BBMD requests with Result messages. But according to the second issue (#167), we agreed that foreign application should reply to unicast requests even if no register response has been received.

Combining both of these ticket results, I found an issue in my previous patch regarding treatment of BBMD requests for foreign application which has not already received the registration confirmation.
BBMD requests are ignored because the `BIPForeign.confirmation` checks the registration validity before processing of BBMD requests. Then I suggest you to move the part regarding registration validity at the end of the `confirmation` method.

[0021_UnicastForBBMDRequest.txt](https://github.com/JoelBender/bacpypes/files/1836725/0021_UnicastForBBMDRequest.txt)

JoelBender, 2018-05-24:
If the foreign device is allowed to communicate without being registered, then the only packets that should dropped are `ForwardedNPDU` because the message could have been intended for a foreign device application instance using the same socket.  In all cases `OriginalBroadcastNPDU` should be silently dropped because maybe the application happened to pick a random socket that corresponds to a network.

JoelBender, 2018-05-24:
Note that all the tests in `tests/test_bvll/test_foreign.py` are positive tests for good behavior, there aren't any tests for the functionality that was added in #166, or this new change.

closed 2018-05-25 by JoelBender