JoelBender/bacpypes #178: BBMD behind a proxy.

amrx101, 2018-03-27:
Run BBMD behind a proxy. The public IP of the proxy should be used by the Foreign Device to register and communicate. [Document](http://www.bacnet.org/Addenda/Add-135-2008o.pdf).

Setup 
[image]

JoelBender, 2018-03-29:
I was expecting that BR1, BR2, and BR3 would be using the same port for network 1 traffic and network 4 traffic and therefore it would have all kinds of work to do.  But the only traffic they should be receiving is Forwarded NPDU messages from their peers and foreign device traffic from FD2, original unicast and original broadcast messages they might get on port 47809 should be dropped or rejected.

Unlike regular BBMDs like B1 which are initialized with their directly associated IP address and port, the BIP layer of BR1, BR2, and BR3 are initialized with their "global" address and their BDTs are all full of global addresses.  The FDT will contain global addresses as well, like FD2 that sits on the Internet and registers with the public address of BR3, or the unlucky FD1 that registers with BR1 as a foreign device.

In addition to dropping original unicast and original broadcast messages, it would probably be a good idea to drop BDT and FDT maintenance traffic from outside the corporate intranet, so that would mean adding some kind of accept/reject rules.  And while we're at it, apply the same set of rules for incoming foreign device traffic to limit it to known devices.

This functionality is very close to (but subtly different from) the `BIPBBMD` class, so I'm proposing a new `BIPNAT` class.  I'll also have to build an `IPNATRouter` in the vlan module for testing . . . _no more code without being accompanied by tests, right?!_

amrx101, 2018-03-30:
@JoelBender , I did a minor hack and got it working in my local setup with `BBMD` behind a` Corporate VPN` and FD out in the open. Essentially I had the `ForwardedNPDU` have the `global address` and not the 'self.bbmdAddress` as per the document.

` Except when propagating a received Forwarded-NPDU message, the "B/IP Address of Originating Device" field in
Forwarded-NPDU messages is the global IP address and port of the NAT Router through which the BBMD
communicates. This is required so that responding devices on the remote subnet may communicate with the originating
device. Received Forwarded-NPDU messages are propagated as-is to foreign devices and to the local IP subnet as
defined in J.4.5 `

The approach you suggest with a separate new class makes sense. Also +1 for tests. If you need any help with tests please do ping me in comments.

RorschachRev, 2018-05-06:
Have you considered following an implementation of the RFC3489 specifically designed to handle this use case? https://github.com/jtriley/pystun This includes NAT, and is only 65 lines of code.

JoelBender, 2018-05-06:
I haven't heard of STUN before, it seems like it would help with the administration of BR1 and BR2 to know how they are mapped to the external world, but it seems like it still requires port forwarding to be configured, and both the BACnet Virtual Link Layer and Network Layer can have IP addresses buried in them.  Thanks for the reference, and it seems that there is a [public STUN server list](https://gist.github.com/mondain/b0ec1cf5f60ae726202e), at least one of those could be for exploring.

RorschachRev, 2018-05-08:
The main advantage of STUN is that RFC 5389 is battle tested and works in the real world with tens of thousands of pieces of software against thousands of pieces of hardware.
If you want me to implement anywhere, I'll be happy to help.

JoelBender, 2018-06-04:
This branch has been brought up to date as if it is going to be included in the 0.17.2 release.  For tests it will need a scaffolding like the diagram and it would be nice if it could use the same tests for BBMDs.

JoelBender, 2018-07-04:
This in the latest release (0.17.3), open a new issue if there are bugs.

@RorschachRev it would be interesting to use STUN and see how well it works, but it is not part of the current standard.  I would expect that everyone's efforts (including mine) are going to be focused on the new BACnet/SC (websockets over TLS).  If you think this would help the BACpypes community even without the support of the HVAC vendors hack away!

closed 2018-07-04 by JoelBender