JoelBender/bacpypes #179: Schedule Reading

clafouti2104, 2018-03-29:
Hello,

I'm trying to read the schedule of my bacnet device using the MultipleReadProperty.py. I receive a DailySchedule Array but I don't know how to read/decode it. Can you help me please?

Thank you!

Here's my command:
`point_list = [
    (<DEVICE_IP_ADDRESS>, 'schedule', 1, 'weeklySchedule'),
]
`
Here's my results:
`((<DEVICE_IP_ADDRESS>, 'schedule', 1, 'weeklySchedule'), [<bacpypes.basetypes.DailySchedule object at 0xb6d925ac>, <bacpypes.basetypes.DailySchedule object at 0xb6d9256c>, <bacpypes.basetypes.DailySchedule object at 0xb6d92d2c>, <bacpypes.basetypes.DailySchedule object at 0xb6d92dec>, <bacpypes.basetypes.DailySchedule object at 0xb6d92e8c>, <bacpypes.basetypes.DailySchedule object at 0xb6d92f2c>, <bacpypes.basetypes.DailySchedule object at 0xb6d92fcc>])
<bacpypes.basetypes.DailySchedule object at 0xb6d925ac>
<bacpypes.basetypes.DailySchedule object at 0xb6d9256c>
<bacpypes.basetypes.DailySchedule object at 0xb6d92d2c>
<bacpypes.basetypes.DailySchedule object at 0xb6d92dec>
<bacpypes.basetypes.DailySchedule object at 0xb6d92e8c>
<bacpypes.basetypes.DailySchedule object at 0xb6d92f2c>
<bacpypes.basetypes.DailySchedule object at 0xb6d92fcc>`

JoelBender, 2018-03-30:
First, think of `Sequence` like a `struct` that has defined fields, some of which might be optional.  Then think of `SequenceOf` like a Python list.  In Python terms, the weekly schedule is a list of daily schedules, one for each day.  The `DailySchedule` object has one attribute, a `daySchedule`, which is a list of `TimeValue` objects.  The `time` attribute is going to be the time of day that the schedule value changes from whatever it was to the `value` attribute.

For example, this daily schedule says that on some day the schedule will have its default value until 8:00am when it changes to 8, then goes back to its default value until 5:00pm when it changes to 42, then goes back to default at 7:00pm:

    >>> from bacpypes.primitivedata import Integer, Null
    >>> from bacpypes.basetypes import DailySchedule, TimeValue
    >>> 
    >>> x = DailySchedule(
    ...     daySchedule=[
    ...     TimeValue(time=(8,0,0,0), value=Integer(8)),
    ...     TimeValue(time=(14,0,0,0), value=Null()),
    ...     TimeValue(time=(17,0,0,0), value=Integer(42)),
    ...     TimeValue(time=(19,0,0,0), value=Null()),
    ...     ])

These objects have a `debug_contents()` method, so you can call it on each of the seven daily schedules you have been given:

    >>> x.debug_contents()
        daySchedule
            [0]            time = (8, 0, 0, 0)
                value = <bacpypes.primitivedata.Integer object at 0x7f01ae7d32b0>
            [1]            time = (14, 0, 0, 0)
                value = <bacpypes.primitivedata.Null object at 0x7f01af9272b0>
            [2]            time = (17, 0, 0, 0)
                value = <bacpypes.primitivedata.Integer object at 0x7f01af921898>
            [3]            time = (19, 0, 0, 0)
                value = <bacpypes.primitivedata.Null object at 0x7f01ae6cbe80>
    >>> 

And if that seems a little too obscure, it also has a `dict_contents()` method:

    >>> x.dict_contents()
    {'daySchedule': [{'time': (8, 0, 0, 0), 'value': 8}, {'time': (14, 0, 0, 0), 'value': ()},
    {'time': (17, 0, 0, 0), 'value': 42}, {'time': (19, 0, 0, 0), 'value': ()}]}

> _Warning: The dictionary returned by `dict_contents()` is likely to change to JSON-LD sometime in the future._

clafouti2104, 2018-04-03:
Thank you, it clear now!
It works perfectly

closed 2018-04-03 by clafouti2104