JoelBender/bacpypes #180: Property Writing

clafouti2104, 2018-04-03:
Hi everybody,

I'm trying to create a script which writes a property but I got some errors. Sorry but I'm a really bad python developper.
I take the ReadWriteProperty.py and try to adapt it (I don't want to have a console, the script will write a single property).
Here's my code
`#!/usr/bin/env python

"""
This application presents a 'console' prompt to the user asking for commands.

For 'read' commands it will create ReadPropertyRequest PDUs, then lines up the
coorresponding ReadPropertyACK and prints the value.  For 'write' commands it
will create WritePropertyRequst PDUs and prints out a simple acknowledgement.
"""

import sys

from bacpypes.debugging import bacpypes_debugging, ModuleLogger
from bacpypes.consolelogging import ConfigArgumentParser
from bacpypes.consolecmd import ConsoleCmd

from bacpypes.core import run, enable_sleeping
from bacpypes.iocb import IOCB

from bacpypes.pdu import Address
from bacpypes.object import get_object_class, get_datatype

from bacpypes.apdu import SimpleAckPDU, \
    ReadPropertyRequest, ReadPropertyACK, WritePropertyRequest
from bacpypes.primitivedata import Null, Atomic, Integer, Unsigned, Real
from bacpypes.constructeddata import Array, Any

from bacpypes.app import BIPSimpleApplication
from bacpypes.service.device import LocalDeviceObject

# some debugging
_debug = 1
_log = ModuleLogger(globals())

# globals
this_application = None

#
#   ReadWritePropertyConsoleCmd
#

@bacpypes_debugging
#
#   __main__
#

def main():
    global this_application

    """write <addr> <type> <inst> <prop> <value> [ <indx> ] [ <priority> ]"""
    # parse the command line arguments
    args = ConfigArgumentParser(description=__doc__).parse_args()
    try:
        addr = '192.168.10.50'
        obj_type = 'analogValue'
        # obj_type = int('0')
        obj_inst = int('7')
        prop_id = 'presentValue'
        priority = int('8')
        value='2250'
        indx = None
        # get the datatype
        datatype = get_datatype(obj_type, prop_id)

        if _debug: _log.debug("initialization")
        if _debug: _log.debug("    - args: %r", args)

        # make a device object
        this_device = LocalDeviceObject(
            objectName=args.ini.objectname,
            objectIdentifier=int(args.ini.objectidentifier),
            maxApduLengthAccepted=int(args.ini.maxapdulengthaccepted),
            segmentationSupported=args.ini.segmentationsupported,
            vendorIdentifier=int(args.ini.vendoridentifier),
            )

        # make a simple application
        this_application = BIPSimpleApplication(this_device, args.ini.address)

        # get the services supported
        services_supported = this_application.get_services_supported()
        if _debug: _log.debug("    - services_supported: %r", services_supported)

        # let the device object know
        this_device.protocolServicesSupported = services_supported.value

        if _debug: _log.debug("after services")

        # change atomic values into something encodeable, null is a special case
        if (value == 'null'):
            value = Null()
        elif issubclass(datatype, Atomic):
            if datatype is Integer:
                value = int(value)
            elif datatype is Real:
                value = float(value)
            elif datatype is Unsigned:
                value = int(value)
            value = datatype(value)
        elif issubclass(datatype, Array) and (indx is not None):
            if indx == 0:
                value = Integer(value)
            elif issubclass(datatype.subtype, Atomic):
                value = datatype.subtype(value)
            elif not isinstance(value, datatype.subtype):
                raise TypeError("invalid result datatype, expecting %s" % (datatype.subtype.__name__,))
        elif not isinstance(value, datatype):
            raise TypeError("invalid result datatype, expecting %s" % (datatype.__name__,))
        # if _debug: ReadWritePropertyConsoleCmd._debug("    - encodeable value: %r %s", value, type(value))
        if _debug: _log.debug("    - encodeable value: %r %s", value, type(value))

        # build a request
        request = WritePropertyRequest(
            objectIdentifier=(obj_type, obj_inst),
            propertyIdentifier=prop_id
            )
        request.pduDestination = Address(addr)

        # save the value
        request.propertyValue = Any()
        try:
            request.propertyValue.cast_in(value)
        except Exception as error:
            if _debug: _log.debug("WriteProperty cast error: %r", error)
            # ReadWritePropertyConsoleCmd._exception("WriteProperty cast error: %r", error)

        # optional array index
        if indx is not None:
            request.propertyArrayIndex = indx

        # optional priority
        if priority is not None:
            request.priority = priority

        # if _debug: ReadWritePropertyConsoleCmd._debug("    - request: %r", request)
        if _debug: _log.debug("    - request: %r", request)

        # make an IOCB
        iocb = IOCB(request)
        # if _debug: ReadWritePropertyConsoleCmd._debug("    - iocb: %r", iocb)
        if _debug: _log.debug("    - iocb: %r", iocb)

        # give it to the application
        this_application.request_io(iocb)
        if _debug: _log.debug("    - run request")

        # wait for it to complete
        iocb.wait()
        if _debug: _log.debug("    - after wait")

        # do something for success
        if iocb.ioResponse:
            if _debug: print("    - response")
            # should be an ack
            if not isinstance(iocb.ioResponse, SimpleAckPDU):
                # if _debug: ReadWritePropertyConsoleCmd._debug("    - not an ack")
                if _debug: print("    - not an ack")
                return

            sys.stdout.write("ack\n")

        # do something for error/reject/abort
        if iocb.ioError:
            if _debug: _log.debug("    - ioError")
            sys.stdout.write(str(iocb.ioError) + '\n')
            if _debug: _log.debug("  after  - ioError")
    except Exception as error:
        if _debug: _log.debug("exception: %r", error)
        
    _log.debug("running")

    run()

    _log.debug("fini")

if __name__ == "__main__":
    main()
`

Here's the results
`root@domaboxdev:/home/manager/bacpypes/samples# python ReadWriteProperty.py --debug --ini ./BACpypes.ini
DEBUG:__main__:initialization
DEBUG:__main__:    - args: Namespace(buggers=False, color=False, debug=['__main__'], ini=<class 'bacpypes.consolelogging.ini'>)
DEBUG:__main__:    - services_supported: <bacpypes.basetypes.ServicesSupported object at 0xb6d479ac>
DEBUG:__main__:after services
DEBUG:__main__:    - encodeable value: <bacpypes.primitivedata.Real object at 0xb6d47a0c> <class 'bacpypes.primitivedata.Real'>
DEBUG:__main__:    - request: <bacpypes.apdu.WritePropertyRequest(15) instance at 0xb6d47a2cL>
    <bacpypes.apdu.WritePropertyRequest(15) instance at 0xb6d47a2cL>
        pduDestination = <Address 192.168.10.50>
        pduExpectingReply = 1
        pduNetworkPriority = 0
        apduType = 0
        apduService = 15
        objectIdentifier = ('analogValue', 7)
        propertyIdentifier = 'presentValue'
        propertyValue
                <bacpypes.primitivedata.Tag(real) instance at 0xb6d3b6cc>
                    tagClass = 0 application
                    tagNumber = 4 real
                    tagLVT = 4
                    tagData = '45.0c.a0.00'
        priority = 8
        pduData = x''
DEBUG:__main__:    - iocb: <bacpypes.iocb.IOCB(1) instance at 0xb6d3b68c>
    <bacpypes.iocb.IOCB(1) instance at 0xb6d3b68c>
        args = (<bacpypes.apdu.WritePropertyRequest(15) instance at 0xb6d47a2cL>,)
        kwargs = {}
        ioState = 0
        ioComplete = <threading._Event object at 0xb6d3b50c>
        ioCallback = []
        ioPriority = 0
DEBUG:__main__:    - run request
DEBUG:__main__:    - after wait
DEBUG:__main__:    - ioError
no task manager
DEBUG:__main__:  after  - ioError
DEBUG:__main__:running
`
I don't know why I'm having the "no task manager" error.

Thank you

JoelBender, 2018-04-06:
The "core" of BACpypes has things that need to happen at a specific time, like re-sending a request if it didn't see a response or updating a schedule to a new value, and doing things on a regular basis like clearing out cache entries.  These things are "tasks" and organizing the list of tasks is up to an instance of a `TaskManager` all of which are in the `bacpypes.task` module.

There is only one task manager (an instance of the `TaskManager` class) that is shared throughout the library and because it follows the "singleton pattern" whenever you call the class to build an instance you'll always get the same one:
```
>>> from bacpypes.task import TaskManager
>>> tm1 = TaskManager()
>>> tm2 = TaskManager()
>>> tm1 is tm2
True
```
For the truly adventurous, BACpypes has a provision for you to provide your own task manager by subclassing and creating your instance before the rest of the library needs one and the library will use yours instead:
```
>>> from bacpypes.task import TaskManager
>>> class MyTaskManager(TaskManager):
...     pass
... 
>>> tm1 = MyTaskManager()
>>> tm2 = TaskManager()
>>> tm1 is tm2
True
```
So the reason you are getting the "no task manager" error is because your application is attempting to schedule a task (in this case, what to do if you don't get a response to your request) and one hasn't been created, but it will be created auto-magically by the `run()` function.  So whatever you do you want to make sure the functions get called after the core is up and running, and by calling `deferred()` it will put a function in the list of tasks to execute.

The more difficult problem with your application is that you are using the threaded version of the IOCB functions (specifically `iocb.wait()`) which is based on events. With only one thread the event will never get set so your application will stall.  The solution is to have your WritePropertyRequest running in it's own thread or to add a callback to the IOCB with a function that gets called when the request is complete.

I'm attaching two different versions of your application, look them over and let me know if one fits your way of doing things better than the other or if you have more questions.
[ReadWriteProperty-ans.zip](https://github.com/JoelBender/bacpypes/files/1884217/ReadWriteProperty-ans.zip)

clafouti2104, 2018-04-07:
Thank you for your answer.
It's more clear now.
I got a last question, I want too have a tcp server which listens for write requests to send. So I try to adapt to adapt your second example, but I cannot close my thread client. So the first request sends the write command, the second sends 2 requests, the third sends 3 times and so on...
Can you help me please?
[scripts.zip](https://github.com/JoelBender/bacpypes/files/1886353/scripts.zip)

JoelBender, 2018-04-07:
Ah, there's nothing quite like debugging threaded applications and sockets, which is why I avoid them and the _raison d'être_ for the `bacpypes.iocb` module.  There are even more wonderful idiosyncrasies  lurking in TCP connection management, so the `bacpypes.tcp` module simplifies those.  I mashed together the `TCPServer.py` sample code and the stuff you'll recognize, see attached.

[WritePropertyTCPServer.zip](https://github.com/JoelBender/bacpypes/files/1886607/WritePropertyTCPServer.zip)

Seikodi, 2018-05-21:
Hi Joel and bonjour clafouti ;-) .
I'm quite new to python, bacnet and by the way bacpypes :-) .
Like clafouti I want to create a script that I can use to write a property but this way.

>import  WriteProperty
>ansWrite = WriteProperty.write(argu)
>>if ansWrite != 'ack'
>>print 'erreur d ecriture'

I've done this script inspired by Joel work but can't get the expected result, can you help me please?
[WriteProperty.zip](https://github.com/JoelBender/bacpypes/files/2023130/WriteProperty.zip)

closed 2018-06-21 by clafouti2104