JoelBender/bacpypes #183: Proposal: Change representation of bacpypes.pdu.Address to handle any port

pshirali, 2018-04-14:
Hello Joel,

The `__str__` of `bacpypes.pdu.Address` has an implementation that prints the entire address in hex when a non-standard BACnet port is being used.

1. Could you please let me know if this has any particular significance?

1. Would it be a problem if bacpypes printed the Address in human-readable form for _any_ port?

**Usecase:**
When bacpypes is used in an environment which involves port-forwarding (using ports outside of BACnet), the Address repr is not readable. When I run bacpypes with debugging enabled, the `debug_contents` of the PDUs show the addresses for `pduSource` etc in hex form.

This becomes especially difficult when I'm trying to ascertain the contents of BACnet packets while also trying to debug/validate my routing or NAT rules. Having bacpypes output the actual IP and port helps a lot in such cases.

Following is the proposed code change (as done under py27):

```
diff --git a/py27/bacpypes/pdu.py b/py27/bacpypes/pdu.py
index eb848bd..39814e2 100755
--- a/py27/bacpypes/pdu.py
+++ b/py27/bacpypes/pdu.py
@@ -302,10 +302,9 @@ class Address:
                 rslt += str(ord(self.addrAddr))
             else:
                 port = struct.unpack('!H', self.addrAddr[-2:])[0]
-                if (len(self.addrAddr) == 6) and (port >= 47808) and (port <= 47823):
+                if (len(self.addrAddr) == 6):
                     rslt += '.'.join(["%d" % ord(x) for x in self.addrAddr[0:4]])
-                    if port != 47808:
-                        rslt += ':' + str(port)
+                    rslt += ':' + str(port)
                 else:
                     rslt += '0x' + btox(self.addrAddr)
             return rslt
@@ -319,10 +318,9 @@ class Address:
                 rslt += str(ord(self.addrAddr[0]))
             else:
                 port = struct.unpack('!H', self.addrAddr[-2:])[0]
-                if (len(self.addrAddr) == 6) and (port >= 47808) and (port <= 47823):
+                if (len(self.addrAddr) == 6):
                     rslt += '.'.join(["%d" % ord(x) for x in self.addrAddr[0:4]])
-                    if port != 47808:
-                        rslt += ':' + str(port)
+                    rslt += ':' + str(port)
                 else:
                     rslt += '0x' + btox(self.addrAddr)
             return rslt
```

The difference in output is as follows:

**Before:**
```
>>> from bacpypes.pdu import Address
>>> a = Address("192.168.1.2/24:12345")
>>> a
<Address 0xc0a801023039>
```

**After:**
```
>>> from bacpypes.pdu import Address
>>> a = Address("192.168.1.2/24:12345")
>>> a
<Address 192.168.1.2:12345>
```

Could you please provide your thoughts on this change?
If this is acceptable to you, could I make a PR with the above change (for py25, py27 and py34)?

Cheers,
Praveen

JoelBender, 2018-04-15:
In its raw form the BACnet address is just an octet string.  If it's one octet long the most common use is for MS/TP and ARCNET addresses, an unsigned integer is really common.  Back in the days when BACnet/IP was first introduced, Ethernet address were the most common long version, and the regular way of displaying addresses separated by colons looked awkward with the colon as the separator between the network portion (so 12:03:04:23:a3:2c:fe is hard to see that 12 is the network and it is _not_ in hex).  There are lots of gateways that use the 6-octet form for virtual MAC address.

I was expecting that when it came time to support IPv6 that I would use something like [fe80::d48d:3439:32ed:9bc9/64] so that the network number, address, and port can all still be separated by colons (the first option of section 6 in [RFC 5952](https://tools.ietf.org/html/rfc5952)).  There are also the display of VMAC addresses (3-octets long) where the simplest way is just a decimal string because it matches the device identifier, but for small device identifiers there is no way to tell if 231 should be one or three octets long (see Section H.7.2).

I feel your pain with NAT rules and the fact that all of those addresses are IPv4, so how about a new subclass of `LocalStation` called `IPv4Address` that accepts of tuple (and would use `pack_ip_addr`) or a 6-octet string and always returns the common string format.  In the `udp`, `bvll`, and `bvllservice` modules it would use this new class as a replacement for `Address`.

It's a significantly larger change than what you are proposing, but it could help set a precedent for the time IPv6 and VMAC addresses are implemented.

JoelBender, 2018-04-15:
It occurred to me that if you are dealing with NAT you are probably also dealing with foreign devices, so you might want to check out the `181-foreign-device-invalid-pdu-type` branch, which is waiting for me to revamp the BVLL tests before releasing it.  It also includes a `NATRouter.py` sample application that is intended to work according to **J.7.5 BBMD Operation with Network Address Translation**, but it also needs unit tests.

While it was simple enough to create a virtual IPv4 intranet for testing and the `IPRouter`, simulating a NAT router with forwarding rules could be tricky, leading to more testing.  Ah, the tangled web we weave!

pshirali, 2018-04-17:
Please correct me if I'm wrong. From your first comment, what I understood is that the `bacpypes.pdu.Address` is agnostic, and there is a possibility that a 6-octet string could be assumed to represent not-just an IPv4 address (one being VMAC). In an incoming 6-octet string, it'd be an educated guess that if the last two octets (port) happened to be within the range of BACnet standard ports, then we'd represent the entire address as an IPv4 address?

Could the address family be an attribute on the `Address` class?. We'd set `Address.addrFamily` to a flag that represents IPv4, IPv6, VMAC etc, such that there's an additional flag that can be relied upon when representing the address (and not just number of octets) in the `__str__` code.

Now, when we compose an address ourselves, it'd be easy to set the `addrFamily` attribute based on the format of the input, as we'd expect it to match one of the regexes in `Address.decode_address`. This being an alternative to explicitly composing using `IPv4Address` which would pack the inputs. I don't have a problem using a class `IPv4Address` to compose the underlying `Address`, whilst also setting `addrFamily` along the way. But, this could only work if I know beforehand that I'm dealing with an IPv4 address. I'd however assume that a majority of the address composition would happen using the `Address` class directly in the bottom layers?

_In the udp, bvll, and bvllservice modules it would use this new class as a replacement for Address_

I'm still a little confused about this aspect. I'd assume that we'd be able to compose an Address by calling Address(<incoming_value>). Currently, since we are dealing with IPv4, we call `Address(unpack_ip_addr(bvlpdu.get_data(6)))` in BVLL. Could you let me know your views on how you'd plan to deal with IPv6 here? (Would it be a combination of bvll + AnnexU binding, and the binding of AnnexJ vs AnnexU would decide how many bytes to fetch into `unpack_ip_addr`)?

Yes, I am dealing with NAT and foreign devices :grin:. I will checkout your `181-foreign-device-invalid-pdu-type` branch soon.

JoelBender, 2018-04-18:
First, the `addrFamily` idea is very appealing, thank you!

The 6-octet length is ambiguous between Ethernet and BACnet IPv4 addresses, and yes, it just makes a guess that when the last two octets happen to be BACnet UDP sockets the first four are probably IPv4 address.  You are also correct that the address instance construction comes from the bottom layers going upstream and in those cases setting the family is easy.  When the addresses come from the top of the stack going downstream they are quite often from a database or configuration file and represented as strings so I wanted the parsing code to be as flexible as possible without being ambiguous.

I _was_ thinking that the `IPv4Address` would absorb the parsers out of the `Address` class, and now I think that might be too much.  An IPv6 address tuple `('fe80::32ed', 47808)` needs to be distinguished from an IPv4 address tuple `('1.2.3.4', 47808)` so maybe those two can be specialized subclasses and set the address family as you suggested.  To follow the current design pattern, an IPv6 `addrAddr` octet string would be 18 octets long, the first 16 are the 128-bit IPv6 address and the last two as the port.

Because the BVLL encoding and decoding is quite different, not only is the header a different size but there are more PDU types and the PDU type code for the same "function" is different (like `OriginalBroadcastNPDU`).  So I was anticipating moving the `udp`, `bvll`, and `bvllservice` modules into a subpackage called `ipv4` and then building a new subpackage called `ipv6` with similar modules.  The existing modules would turn into the python version of "look over there" in one version, then have a deprecation warning, then be pulled out.  The two different UDP modules, `bacpypes.ipv4.udp` and `bacpypes.ipv6.udp` would have two different `unpack_ip_addr()` functions.

Packets going upstream from the IPv6 UDPDirector to the corresponding BVLL service objects would have these huge octet strings and I'm assuming that all of the VMAC binding happens at that layer and going upstream from that to the network layer is the 3-octet VMAC address.  What gets presented at the application layer is an open question, I'm guessing that the VMAC address for a device will be stored as an additional attribute in the `DeviceInfo` object that will be shared between the application and BVLL layers.

Maybe I'm wandering too far off track for this issue, but I will poke at this `addrFamily` concept some more.

pshirali, 2018-04-18:
Since bacpypes is currently dealing only with IPv4 when presented with a 6-octet address, would it be safe to get the `Address` to represent `IPv4:port`? In other words, do you feel there is still a chance of ambiguity in its present form if we went ahead with the change mentioned in this issue? Please let me know if my proposed change could be made, or we should hold/reject.

----

On the larger topic of handling addresses for all families and breaking the parsing responsibilities, here's one approach to deal with it. Please feel free to consider it if you think it is suitable.

1. The baseclass `Address` can be built with any address input and would have a hex octet representation for itself `<Address 0x...>`.
1. The `Address` would further get subclassed into their specific families, with multiple hierarchies, like `MSTPMACAddress`, `EthernetAddress`, `IPAddress`, breaking further down into `IPv4Address`, `IPv6Address` etc.
1. The baseclass `Address` will have a _classmethod_ called `build` (for example), who would consume an input and dish out the address in its most specialized class. This would be done by handling the input parsing only at its own layer, and then engaging a subclass's `build` method on its way down.
1. To handle conflicts when dealing with addresses of same length, the build could also take a kwarg called `family`, which will explicitly override any logic and cast the address as per the supplied family.

Example:
```
>>> addr = Address.build(<18-octets>)
>>> print(addr)
<IPv6Address [<16-octets>]:<port>>

Internally, the flow would be Address.build()->IPAddress.build()->IPv6Address.build()

>>> addr = Address.build(<6-octet>)
>>> print(addr)
<Address 0x...>        # as there is more than one repr, casting has to be explicit

>>> addr = Address.build(<6-octet>, family=address.IPv4)     # address.IPv4 being some enum
>>> print(addr)
<IPv4Address <a.b.c.d>:<port>>

>>> addr = Address.build(<6-octet>, family=address.Ethernet)
>>> print(addr)
<EthernetAddress ab:cd:ef:01:02:03>
```

For packets coming in from the bottom layers, the `family` could be fed in based on the interface & address to which the stack is bound.

The functionality of `pack_ip_addr` and `unpack_ip_addr` would belong somewhere within the Address subclasses and need not be used in the address building process (may be for debugging). BVLL now has a way of directly consuming octet-string and need not call these functions. It'd instead just call `Address.build(<bvlpdu.get_bytes(..), family=<family_hint>)`

I don't have any particular reservations around how the ipv4 and ipv6 code would be structured. However, it'd be great if those submodules implemented derived classes with varying functionality (as much as possible), as opposed to those submodules being package clones (with copies of entire files). This is largely to avoid fixing common code issues in multiple subpackages.

It'd also be great if the application building process took care of binding the appropriate lower layers using may be the application's address as a hint (such that users would never have to explicitly know/import anything from bacpypes.ipv4.udp etc).

JoelBender, 2018-04-18:
I like the direction this is going, so this is "accepted" with the details to be ironed out.  In addition to IPv6 and BACnet/SC (_BACnet Secure Connect_, which hopefully will be voted out for public review today) there is the`NetworkPortObject` which comes with `NetworkType` enumeration (called BACnetNetworkType in the standard).  Renaming `addrFamily` to `addrNetworkType` and synchronizing the enumeration values seems like a nice fit.

Your `build()` class method, following the factory method design pattern, is a perfect application for a bit of metaclass programming.

JoelBender, 2018-05-26:
This branch has been brought up to date with the (still pending but about to be released) 0.17.1, passes all the tests.

JoelBender, 2021-09-02:
This has been rolled into **bacpypes3** and extended in various ways, so I'm closing this issue and delete the related branch.  Thank you Praveen, wherever you are, and say hi to Mrs. Calabash.

closed 2021-09-02 by JoelBender