JoelBender/bacpypes #184: Addressing a slave device

tomharvey, 2018-05-01:
I've had a read through https://github.com/JoelBender/bacpypes/issues/77 and it seems tantalisingly close to where I want to be.

I've got a router on ip 10.0.0.1 (for example) and is device ID 10 on BACnet network 100, there is a slave on this Device ID 20 which is on network 200.

I can read points from the master device (ID 10) using:

```
request = ReadPropertyRequest(
    objectIdentifier=('analogValue', 1),
    propertyIdentifier='presentValue',
)
request.pduDestination = Address('10.0.0.1')
```

But I want to be able to make a request for analog value 1 on device ID 20, on the different network - through the same IP address (10.0.0.1)

Do I set the BACnet network of 200 when instantiating my `LocalDeviceObject` or, do I specify the device ID in the `ReadPropertyRequest` object?

Thanks for any pointers you can give.

JoelBender, 2018-05-02:
First, if this seems like a drawn out answer, please understand that I am an Ent, and I talk a long time to say things and I don't say anything unless I can take a long time to say it.  :smile: 

There are three pieces of knowledge to keep in mind about communicating between BACnet devices; the device's identity, the network that it lives on, and the MAC address of the device on its network.  Knowing the device identity there are services (Who-Is and I-Am) that allow you to find its BACnet address, the other two pieces.  You have given me two of the three pieces of this device, the device ID is 20 and the network is 200, but you don't know the MAC address of the device on its network.

BACnet is a little different than other other protocols, when you are communicating with another device and you are both on the same network, the address has no network number.  This is similar to not needing an area code for a phone number.  You only need the BACnet network number when there is a router between the network you are on and the network that the other device is on.  BACpypes calls these "local stations" and "remote stations".

So first find the complete BACnet address of the remote device by running the `DeviceDiscovery.py` sample application and at the prompt enter `whois 20 20` to find all the devices on your intranet with device identifier between 20 and 20.  You will get one response and it will look something like "20 is at 200:99 named snork".  The **99** is the MAC address of the device on its network, the **200** is its network, and **snork** is the object name of its device object, a.k.a., the device name.

The `Address` object takes all kinds of formats for addresses and tries to be smart about the kind of address you want.  `Address("1.2.3.4")` is an IPv4 address on the local network, `Address("200:1.2.3.4")` is an IPv4 address on an IP network other than your own, etc.  In this case I've assumed that the device you are talking to is on an MS/TP or ARCNET network that uses very small, single octet address, and **99** is its address.

    $ python
    Python 2.7.12 (default, Dec  4 2017, 14:50:18) 
    [GCC 5.4.0 20160609] on linux2
    Type "help", "copyright", "credits" or "license" for more information.
    >>> from bacpypes.pdu import Address
    >>> x = Address("200:99")
    >>> x
    <Address 200:99>
    >>> x.addrNet
    200
    >>> x.addrAddr
    'c'

Note that the address is bytes, not a string, it just looks like a string in Python2.  This is the address object that you use in your ReadPropertyRequest.

Your `LocalDeviceObject` contains the device identifier that you want to use for your application when other devices want to talk to you.  The BACnet address of your "device" is a special form called a BACnet/IP address and it is the IP address of the machine you are running the BACpypes application, along with the UDP port number (normally 0xBAC0, 47808).

_Note that you don't know the network number of the network you are on!_  The knowledge of the topology is built into the configuration of the routers, and your application is not a router.  So this device that you are talking to will know because the router will fill in your network number in the network layer header (the SNET/SLEN/SADR fields), just like you'll know its network when the I-Am response comes back in the device discovery process.

(There are new services to discover what network you are on, but they are not available in BACpypes yet.)

tomharvey, 2018-05-02:
Thanks for your excellent reply, and library.

I’m communicating between my bacpypes app and the bacnet device across a VPN - so broadcast will not work; I assume this excludes me from using the discovery and Who-is scripts?

I can certainly get the MAC address of the device through physical access though. Can I check my assumption of the next steps?

 I think I will need to declare the `request.pduDestination = Address()` using that MAC address, the BACnet network ID and the router IP? I will need to read more on how `Address()` does its magic parsing, or make use of the subtypes Can IPv4 and BACnet networks be mixed in Address in this way?

Thanks again.

JoelBender, 2018-05-02:
You are correct, without the ability to be in the same "broadcast domain" as the other devices your Who-Is  broadcast request won't go where you want it to go.  There are two solutions, (1) install a BBMD on the network where the router is located (and there might already be one there) and become a foreign device that registers with the BBMD or (2) use "static" routes.

Starting with (1) assume that there is a BBMD already install on the network.  You can run the `DeviceDiscoveryForeign.py` application which uses a subtly different BACnet/IP Virtual Link Layer (BVLL) which needs the IP address of the BBMD and a time-to-live to maintain the registration.  Your broadcast Who-Is is sent to the BBMD which broadcasts it on your behalf.  The router-to-network traffic is forwarded as well, so the I-Am response comes back as you might expect.

With option (2) there is an additional command in the `ReadProperty.py` called `rtn` and it is used to tell the network layer of your application the address to use for your network 200.  Once the route is established by telling the network layer of your application (the Network Service Access Point) it won't need to do any broadcasts.

There is a *very* remote chance that your network uses IPv4 Class D multicast address *and* your VPN service will span the multicast group across the VPN.  In this *very rare* case, your broadcasts will be forwarded for you by the VPN/router and you don't need to do anything.  There is a better chance this will become more widespread with IPv6.

tomharvey, 2018-05-02:
Thanks for all of your help so far - as you may tell, I've absolutely no problem with your long explanatory answers - finding them very informative.

JoelBender, 2018-05-03:
The tshark dumps are a little strange, I don't recognize the first 44 octets, but the `81 0a xx xx` string is an OriginalUnicastNPDU, which makes sense, and the rest of the packet decodes easily.  Successfully reading from 10.0.0.1 would imply that the device has a BACnet stack and isn't just a router (most BACnet routers have the option of doing more things than just routing, so this makes sense).

Checking the packets that have returned data, the NPCI of the requests are being forwarded to network 2400, address 10.199.3.56, which leads me to believe that 10.0.0.1 is the address of a router to network 2400 and that network is another BACnet/IP network.  I don't think that 10.0.0.1 is a router to network 100 as you have attempted, I think it's 2400.

Please double check your configuration of the 10.0.0.1 device and confirm that there isn't any NAT mucking around in conjunction with your VPN tunnel.  If you have a "monitor port" or "mirror port" on the switch connected to that mystery device (or plugged into a dump hub acting as a "tap") you could capture what it is actually sending and receiving.  Assuming Wireshark can decode the first 44 octets, there might be some interesting content to inspect.  If you can, run tshark with `-o somefile.pcap` and attach it to a comment on this thread, being careful to filter out the non-BACnet traffic!

tomharvey, 2018-05-04:
Any thoughts on what kind of error is being returned, when I see `noResponse` as the return but a `BACnet-APDU Error` response in the packet sniff?

tomharvey, 2018-05-04:
[bacnet_filtered.pcap.zip](https://github.com/JoelBender/bacpypes/files/1975442/bacnet_filtered.pcap.zip)

I'ver been digging a little more into the PCAP. Looking at request 3 and request 4, the difference is that I'm being explicit about Bacnet network address (2400) but IOCB doesn't seem to get the response, it just times out.

I'm increasingly suspicious that this is a separate error.

As for request 5 - this is where I'm asking for data from the slave on network 2454 and I can parse that the error is an `unknown-object` error. I also notice that when specifying the BACnet network, whether 2400 or 2454, the request is always being sent to MAC address `0a:c7:03:38:ba:c0` - should I be explicitly instructing the request to go to the MAC address of the slave device?

JoelBender, 2018-05-04:
There is some very strange behavior here.  You'll notice that you send out one request in [1] but you get two replies in [2] and [3], but they are coming from two different Ethernet addresses?  I'm not familiar with the "Linux cooked capture" header, or the distinction between "Unicast to us (0)" and the "Sent by us(4)".

Next, your request in [4] is from 10.109.3.38 to 10.199.3.56, but the response [5] is going from 10.199.3.56 to 10.109.3.59, and then another response [6] is going from 10.199.3.56 to 10.109.3.38 as it should.

Your router-to-network command is telling the stack that 10.199.3.56 is the router to these other networks and the application is requesting some from a device on one of those networks, send it to this device as a router and it will forward it along.  This is analogous to sending an IP packet to the Ethernet address of a router (which is sadly called a default gateway) and the router figures out how to get it where it needs to go.

The "nice" thing about [13], [16], and [19] is that they were explicitly rejected with an error.  The not so nice thing is that the error that came back was from a different device than the request was sent to!  Notice how in [13] the destination network is 2454 and the address `0a:c7:03:38:ba:c0`, which matches the IP address you put in the console.  *However,* the source address in the response in [14] is also from network 2454, but the MAC address is `10`.  This is why BACpypes is timing out, the source and destination addresses don't line up.

Just for more information, could you please run the `samples\WhoIsRouter.py` application with debugging turned on:

    $ python samples/WhoIsRouter.py --debug
    > irt 10.199.3.56

This will send an `InitializeRoutingTable` request to the device and because we are not providing it with a new routing table, the response coming back will be its current routing table.  Then we can find out which networks like 2454 are on the other side of the router, and also see if the routing tables are scrambled.  At this point I think your read request should look like this:

    > rtn 10.199.3.56 2454
    > read 2454:10 analogValue 38 presentValue

You should immediately get back an error saying that the object doesn't exist (rather than timing out), then you have to figure out what objects are there, and I can help with that once we get some of this other cruft out of the way.

JoelBender, 2018-05-04:
Ah... [Display Filter Reference: Linux cooked-mode capture](https://www.wireshark.org/docs/dfref/s/sll.html)

tomharvey, 2018-05-04:
I think that double response can be explained by NAT - I'm running the packet sniffer on my IPv4 router (which is also the VPN endpoint, and handles NAT). 10.109.3.56 is that NAT device while 10.109.3.38 is the device which is running the BACpypes stack.

The rest makes total sense now as well - the deviceID is on network 2454 is 15 (not 10) and using that knowledge I get a response! :
```
> read 2454:15 analogValue 38 presentValue
100.0
```

Thanks very much for this education in the BACnet stack and how to use BACpypes! Do you have a donation link? I'll try to get a blog post live about my experiences here and will feedback when that's ready.

closed 2018-05-04 by tomharvey