JoelBender/bacpypes #186: Question regarding exception when try to read list property on remote object

calligp, 2018-05-04:
I try to read the propertyList property of my remote device with a script inspired from the sample `ReadPropertyAny`. But I see an exception when I try to decode the `Any` value retrieve in ReadPropertyAck message. This issue seems be linked to the type of this property.

My sample to decode the value is:

```
    datatype = Tag._app_tag_class[value_tag.tagNumber]
    if not datatype:
        raise TypeError("unknown datatype")

    # cast out the value
    # special case for array parts, others are managed by cast_out
    if issubclass(datatype, Array) and (wIocb.ioResponse.propertyArrayIndex is not None):
        if wIocb.ioResponse.propertyArrayIndex == 0:
            value = wIocb.ioResponse.propertyValue.cast_out(Unsigned)
        else:
            value = wIocb.ioResponse.propertyValue.cast_out(datatype.subtype)
    else:
        value = wIocb.ioResponse.propertyValue.cast_out(datatype)
```

I see the same issue when I use the sample directly.
```
$ python ReadPropertyAny.py

> read 10.33.74.201 device 409 propertyList
ERROR:__main__.ReadPropertyAnyConsoleCmd:exception: DecodingError('too many cast components',)
Traceback (most recent call last):
  File "ReadPropertyAny.py", line 99, in do_read
    value = apdu.propertyValue.cast_out(datatype)
  File "/usr/local/lib/python2.7/dist-packages/bacpypes/constructeddata.py", line 1090, in cast_out
    raise DecodingError("too many cast components")
DecodingError: too many cast components
```

Below the detail about algorithm
Received tagNumber : 9
Expected datatype : <class 'bacpypes.primitivedata.Enumerated'>

The received value type is `<bacpypes.constructeddata.Any instance at 0x7f7615e23950>` but contains a `ListOf(PropertyIdentifier)`.

To retrieve the correct datatype to be able to decode the value properly I use the function `get_datatype` from the `objects.py` module using the object type ("device" in my test) and the property ("propertyList").
With this change the decoding of the received value seems be correct.

Do you confirm this change is the good way for my issue ?

JoelBender, 2018-05-04:
There are two different constructs, the `ArrayOf(PropertyIdentifier)` which is what is in the device object:

    >>> bacpypes.object.get_datatype('device', 'propertyList')
    <class 'bacpypes.constructeddata.ArrayOfPropertyIdentifier'>

and you can make your own `ListOf(PropertyIdentifier)` but be aware that that is not one of the types currently being used:

    >>> from bacpypes.basetypes import PropertyIdentifier
    >>> from bacpypes.constructeddata import ListOf
    >>> ListOfPropertyIdentifier = ListOf(PropertyIdentifier)

Building one should be straight forward, the constructor parameter for both `ArrayOf` and `ListOf` are regular Python lists:

    >>> x = ListOfPropertyIdentifier(['presentValue', 'objectName'])
    >>> x.debug_contents()
        [0] = 'presentValue'
        [1] = 'objectName'

And you can cast them in and out of `Any`:

    >>> from bacpypes.constructeddata import Any
    >>> a = Any()
    >>> a.cast_in(x)
    >>> a.debug_contents()
            <bacpypes.primitivedata.Tag(enumerated) instance at 0x7f3931e74410>
                tagClass = 0 application
                tagNumber = 9 enumerated
                tagLVT = 1
                tagData = '55'
            <bacpypes.primitivedata.Tag(enumerated) instance at 0x7f3931e743d0>
                tagClass = 0 application
                tagNumber = 9 enumerated
                tagLVT = 1
                tagData = '4d'

    >>> y = a.cast_out(ListOfPropertyIdentifier)
    >>> y
    ['presentValue', 'objectName']

Because `Any` has no concept of what the top level construct should be, the same tag list can be cast out as an array:

    >>> ArrayOfPropertyIdentifier = ArrayOfPropertyIdentifier(['objectName', 'presentValue'])
    >>> y = a.cast_out(ArrayOfPropertyIdentifier)
    >>> y
    ['objectName', 'presentValue']

What might be confusing, or a non-feature, or both, is that the "cast out" process abstracts out the datatype, similar to casting out a `Real` and a `Double` will get you a Python float with no knowledge of what it was in the encoding.

JoelBender, 2018-05-27:
My answer above might have been to the wrong question!  The `ReadPropertyAny.py` sample application assumes that all of the values coming back are single application tagged primitive values.  If it checked to see if there's more than one in the tag list and that they are all the same type it could return a Python list of them decoded.  In your case they are all enumerated but it doesn't know what kind of enumeration it is, so it can't give you friendly names.

closed 2018-05-27 by JoelBender

calligp, 2018-05-27:
Ok, thanks for your help.
Using the method `get_datatype` from `object.py` with object type and property name, I am able to decode non primitive data.
Do you want my test script to add in your sample ?