JoelBender/bacpypes #189: Multiple devices in one application?

voney, 2018-05-06:
Hi,

I'm trying to figure out how I can create a server that serves multiple device objects.

So say I have a program that polls multiple devices on a different (not bacnet) platform, how can I make my application return whois information and respond to requests for sensors/values on each device?
I suspect I may be trying to emulate a BBMD, but there's not a lot I can see in the docs about it.
Any help or pointers would be greatly appreciated.

Thanks

JoelBender, 2018-05-06:
See the `IP2VLANRouter.py` which presents itself as a BACnet/IP to VLAN router.  Each virtual device on the virtual LAN gets its own device object, and each device has its own list of objects.  Rather than returning a random value for the `presentValue` of the objects you probably want to use the device and object identifiers or names to get the value from some shared memory service, I happen to like [Redis](https://redis.io/).

voney, 2018-05-07:
Thanks that helps! I've been looking over `IP2VLANRouter.py` and I I think I'm slightly confused by the language. In this case the VLAN isn't a traditional networking style VLAN that has to exist on the host, it's just a virtual network within the program? And if that's the case, what kind of values should I be passing to it for it's address and network ID?

JoelBender, 2018-05-07:
You are correct, it's just a virtual network, not an 802.1Q VLAN.  It has to be given a BACnet network number that does not otherwise exist in your BACnet intranet, and the device instance numbers also have to be unique network wide.  If you have other BACnet devices on your network, find out their networks and device instance numbers and find the network administrator that assigns them.  If you are on a test network and your BACnet traffic won't interfere with other devices, you are the administrator!

voney, 2018-06-19:
Hi, I've been playing around with this for a little while and I can't seem to get even the `IP2VLANRouter.py` to respond to WhoIs requests. I'm using Yet Another BACnet Explorer and am able to find the `WhoIsIAm.py` application just fine when using the same setup.

Am I missing something?

JoelBender, 2018-06-20:
The samples might have been out of date with the API changes that have been made recently.  I created a #189 branch and pushed updated *IP2VLANRouter.py* and *WhoIsIAm.py* samples.  For short term testing I ran the router on port 47808 acting as a router between network 5 on the IP side and 6 on the VLAN side, with three virtual devices.  The IP address is my workstation.
```
      $ python3 IP2VLANRouter.py 10.0.1.31 5 6 --count 3 --debug
```
In another window I'm running the WhoIsIAm.py* sample on port 47809.  Because the two are not in the same broadcast domain, I told it the address of the router to network 6 first, then used a directed broadcast to reach the VLAN applications:
```
    $ python3 samples/WhoIsIAm.py --ini BAC1.ini --debug
    > rtn 10.0.1.31 6
    > whois 6:*
```
I will try this again on a different network setup, in the mean time, try checking out and installing the `189-multiple-devices` branch and see if you fare better.  From YABE double check that you are sending global broadcast requests so they get routed, maybe turn on debugging of the UDP layer by giving it a `--debug bacpypes.udp` on the command line and see what it prints out.

JoelBender, 2018-06-20:
The changes have been merged into `stage` for the next release, 0.17.3

JoelBender, 2018-07-04:
Closing the issue, open a new one if you are still having problems.

closed 2018-07-04 by JoelBender

SRikinR, 2022-09-12:
Hey @JoelBender,
Can you please guide me on how to implement "**WhoIsIAm**" and "**WhoHasIIhave**" services **together**? 

I have done WhoIsIAm and am through with it and have also implemented multiple devices in one application but now I want to provide WhoHasIHave Service also. So, Can you please help me with that?!

JoelBender, 2022-09-14:
New issue #468