JoelBender/bacpypes #190: Read all objects or Points(AnalogValue, BinaryValue)

jitendralenka, 2018-05-10:
Hi,

I am searching in 'examples' directory, to read all objects defined in Device.

Would you provide a sample script.

Thanks,
Jitendra Lenka

JoelBender, 2018-05-11:
I added a new sample application [ReadObjectList.py](https://github.com/JoelBender/bacpypes/commit/fc2510cd5d2ed7a38764b065b4d0a43bfd5c4abd) to the `stage` branch so it will be included in the next release.  You can filter out the objects you are not interested in, and rather than reading the object name you can read the `presentValue`.

If you are reading the object list from an MS/TP or ARCNET device that doesn't support segmentation, you'll need to adjust the process of reading the object list to reading the array length and each of the array elements one at a time, which is a bit tedious but not unusual.

jitendralenka, 2018-05-15:
Hi Joel,

Thanks for help and providing script to read objects from device.

Today, this application does not work where multiple devices are connected
to single IP.

But, ReadProperty.py works based on device mac address.

I need a application without console to read data.

I am in need, request to help.

Thanks,
Jitendra lenka

JoelBender, 2018-05-16:
BACnet communications has a one-to-one relationship between a device and a BACnet address, so you cannot have multiple devices connected to a single IP address.  A BACnet IP address includes a UDP port number, so you can run multiple applications on the same device provided you give each one a unique port, but there are some types of messages (unconfirmed broadcast requests) that your applications will not receive unless you do some more work.  I don't thing this is what you are looking to do.

jitendralenka, 2018-06-06:
Hi Joe,

Thanks for reply and apologize for delay as i was trying to modifying
provided code to remove console as it can be fit for my application.

Below is my requirement:

*1. Read all devices (WHOIS)*

Based on script (WhoIsIAm.py), i am able to read all devices, Below is
sample received device info.

<Address 10.9.72.250>
1476
noSegmentation
5
('device', 27000)
<RemoteStation 65001:20>
1024
segmentedBoth
5
('device', 27100)
<RemoteStation 65001:21>
1024
segmentedBoth
5

When i tried to modify script with respect to remove CONSOLE(Avoid
event-driven user input) so that it will be just the application, i could
not able to do it.

*2. Read all objects for a particular device *

Below are sample objects(Points):

('binaryInput', 27)
('binaryInput', 28)
('analogInput', 11)
('analogInput', 12)
('analogInput', 13)

The script called "ReadProperty.py" is able to read all points based on
this argument.

read <Device Id>   <Type>  <Inst>   objectList

read 65001:20    device   27000   objectList

In this case also i could not able to remove CONSOLE to avoid user input.

*3. COV Subscription*

Regarding this, i do not know how to use it and how to get updated value
after subscription.

In all above, i need your help. I need your personal e-mail id to contact.

Thanks,
Jitendra Lenka
Mobile: +91-9916741220
Mysore, INDIA

bh4r4th, 2019-10-15:
Assuming you are using unique ip in the BACnet network you are hooked in. I probably, start something like below:

**1.  Read all devices (WHOIS)**
1. Instantiate Application using BIPSimpleApplication 
2. When you trigger the discoveries, create a iocb based on what time of discovery it is. `Application.request_io(iocb)` will trigger the requests and listen to the callbacks. 
[Reference](https://github.com/JoelBender/bacpypes/blob/4111b8604a16fa2b7f80d8104a43b9f3e28dfc78/samples/WhoIsIAm.py)

**2. Read all objects for a particular device **
1. You need to discover the device object for the device you are interested to discover the points for.
2. In device object you will find the points list
3. Iterate over this list and try reading those points using read_property or read_property_multiple methods depending on your device and segmentation supported.

**3. COV**
1. Change of value I haven't tried. But should be fairly simple if you get 1 and 2. 
2. As a workaround you can do a periodic discoveries and compare them against the initial discovery results in cache. [Not great but at least you can achieve something].

Unfortunately, you can bind single ip to the bacnet device as Joel mentioned above. As a workaround I probably would create a docker cluster network on my machine and spin the device in one docker container and control io using a priority queue. 

Note: This is just my idea on how can it be achieved but I think you can join gitter chat for help from bacpypes community.