JoelBender/bacpypes #191: Write property script

Seikodi, 2018-05-21:
I'm quite new to python, bacnet and by the way bacpypes :-) .
I want to create a script that I can use to write a property but this way.

    import WriteProperty
    ansWrite = WriteProperty.write(argu)
        if ansWrite != 'ack'
        print 'erreur d ecriture'

I've done this script inspired by Joel work but can't get the expected result, can someone help me please?
[WriteProperty.zip](https://github.com/JoelBender/bacpypes/files/2023148/WriteProperty.zip)

Seikodi, 2018-05-25:
Hello Joel thanks for the reply. The application will be running like a function called by a python application just to write to an output from time to time.
It won't be threaded, the purpose is just to write the property at request.

JoelBender, 2020-05-02:
The BACpypes library isn't quite set up for this type of API, please see the [BAC0](https://github.com/ChristianTremblay/BAC0) project.  This may change in the future with bacpypes3 that is Python3.6+ and asyncio based.

closed 2020-05-02 by JoelBender