JoelBender/bacpypes #192: read request no IOCB callback, then COV server fails to emit new values

EtienneHelfer, 2018-05-24:
Hello, 

I have 2 bacnet devices (console application with SPIN 50 ms and enable_sleeping() ), and I often restart those to conduct tests. In parralel to COV subscriptions, I also do 1 second interval polling (read requests on presentValue) for testing. There is no additional threads except the console and core.

Sometimes the COV server stops to work correctly and do not send the change of values.
Another problem that is probably related: When I restart the second bacnet device, the first device sometimes has read request that never get callbacked via IOCB. I do not always get the "noResponse". 

I am using python 2.7 with bacpypes 0.17.0

THanks for the support, 

Best regards,
Etienne

EtienneHelfer, 2018-05-24:
Another problem I noticed: 

Every 10 seconds, I refresh the COV subscription for 15 seconds by re-sending the subscription request, and after some random time it fails to refresh the subscription: eventho the request succeed, I don't get new values from COV server. To make it work again I have to unsubscribe and resubscribe with another process id.

JoelBender, 2018-05-24:
The first thing to try is to run the applications with debugging turned on for the core module, and if the output clutters up the display redirect it to a file like `--debug bacpypes.core.run:somefile.txt` then look in there for exceptions that get captured.  The `run()` function has a catch-all that is good for applications that need to keep running no matter what, but it prevents the application from trace dumping and stopping.  If you debug the entire module it could generate a lot of content.

You might also want to turn on debugging for the `UDPDirector` class which will dump out the packets being sent and received, so when running both applications there is a chance of lining up the outputs of one with the inputs of the other.

So how about running one with  `--debug bacpypes.core.run:program1.txt bacpypes.udp.UDPDirector:program1.txt`, the output of both will go to the same file, and the same on the other program with `program2.txt`.

EtienneHelfer, 2018-05-25:
Hello, thanks a lot for the debug tips.  

I have noticed that the COV  server bug appears when the client renew the subscription. The COV server receives the renew subscription request. It is correctly renewed and the corresponding cov object still exists in the list of subscriptions (app.cov_subscriptions), but the send_cov_notifications (in the COVDetection class) does not get called anymore by the core loop.

I can "deblock" the COV server by canceling the subscription from the client, and register again, but it only works 15 seconds until the subscription renewal. Once again when the subscription is renewed, it still exists in app.cov_subscriptions but the send_cov_notifications does not get called. 

I'll try to find the cause of the bug and post it here, but if you have some ideas / indications it would be great, since I am not familiar with the bacpypes code. 

best Regards,
Etienne

EtienneHelfer, 2018-05-25:
I found that it was stuck in this case :  self.algorithm._triggered. 
I introduced self.algorithm_triggered = False and now it seems to work better.

[image]

Could you propose a better fix or is it fine ?

JoelBender, 2018-06-01:
It might be that easy!  I'm building a test suite for it that follows 135.1 Clause 8.2 ConfirmedCOVNotification Service Initiation Tests, at least some of them, and having tests for multiple clients should be a part of that.

EtienneHelfer, 2018-06-01:
Thank you for the help! 

we are testing it in real project and it seems stable now after this correction. 
Best regards,
Etienne

JoelBender, 2018-06-21:
I'm reopening this issue, applying your patch, and testing.

JoelBender, 2018-06-29:
The last statement of Clause 13.14.2...

> A ConfirmedCOVNotification or UnconfirmedCOVNotification shall be issued as soon as possible after the successful completion of a subscription or re-subscription request, as specified by the 'Issue Confirmed Notifications' parameter.

That code is missing.  The ACK is sent back, but not the notification that follows, so test 9.10.1.1 fails.

JoelBender, 2018-06-30:
Could you please checkout the `192-cov-server-fails` branch and let me know if it works with your application?  Thank you!

The `_triggered` variable is not very well named, it just means that the algorithm has been scheduled to "execute" the next time it gets a change (deferred) and the only way the flag should be cleared is when it actually does execute.  The intent is that when there are multiple changes to an object, like the present value and status flags at the same time, it only executes once.

JoelBender, 2018-07-04:
I'm re-closing this as fixed, please make a new issue if you find any problems.

closed 2018-07-04 by JoelBender