JoelBender/bacpypes #193: blob it is attempting to decode(UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe9 in position 10: invalid continuation byte)

HrishikeshKanabar, 2018-05-28:
Error links
https://files.gitter.im/ChristianTremblay/BAC0/fBXB/Capture-45.
https://files.gitter.im/ChristianTremblay/BAC0/AslA/Capture.JPG

Error
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe9 in position 10: invalid continuation byte

Follow up consversation:

JoelBender
This looks like a character string encoding problem, like the device is saying "this is a utf-8 string" but it's actually a latin-1 string. If you can open a BACpypes issue with the blob it is attempting to decode (at least include the first octet which has the encoding) then I can figure out a way to trap the exception and work around it.

Christian Tremblay @ChristianTremblay May 25 16:53
French weird device ? 0xe9 = é

Hrishikesh Kanabar @HrishikeshKanabar May 25 18:47
yes , this device is in Quebec , that why french further i don't know

Christian Tremblay @ChristianTremblay May 25 19:45
Can you provide vendor name of controller ? Maybe will help if I can test on similar type.
Vendor name and model / controller type.

Hrishikesh Kanabar @HrishikeshKanabar 16:29
Delta Control Model: DAC-1146

https://www.deltacontrols.com/products/hvac-controls/general-purpose-programmable-io-controllers/ms-tp-application-controllers-dac/dac-1146
specs sheet:- http://controlconceptsonline.com/datasheets/delta/DAC-1146.pdf
_

JoelBender, 2018-05-28:
Linux, Python3.5:
```
>>> xtob('e9').decode('utf-8')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe9 in position 0: unexpected end of data

>>> xtob('e9').decode('latin-1')
'é'
```

Linux, Python2.7:
```
>>> xtob('e9').decode('utf-8')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xe9 in position 0: unexpected end of data

>>> xtob('e9').decode('latin-1')
u'\xe9'
```

Curious that these are not the **invalid continuation byte** problem but something else.

JoelBender, 2019-04-25:
Work continuing on #241

closed 2019-04-25 by JoelBender