JoelBender/bacpypes #195: Sending one initial COV notification on COV subscription

EtienneHelfer, 2018-06-01:
Hello, 

I have a bacpypes device COV server, and I would need on new client cov subscription to send to the client an initial cov notification containing at least the presentValue property. 
How could I achieve it simply ? 
I have no problem with modifying the bacpypes cov.py file if necessary. 

Thanks for the support !

Best, 
Etienne

EtienneHelfer, 2018-06-01:
I think I managed to do it by overriding the add_subscription method of the ChangeOfValueServices.
If someone is interested I can share the code

Best,

Etienne

closed 2018-06-01 by EtienneHelfer

JoelBender, 2018-06-05:
Yes please, I'll merge it in, just add it to this issue.  Thank you.

EtienneHelfer, 2018-06-06:
def add_subscription(self, cov):
		ChangeOfValueServices.add_subscription(self,cov)
		#SEND COV AT INIT
		properties_reported = (
			'presentValue',
			'statusFlags',
		)
		current_time = TaskManager().get_time()
		list_of_values = []
		for property_name in properties_reported:
			# get the class
			property_datatype = cov.obj_ref.get_datatype(property_name)
			
			# build the value
			bundle_value = property_datatype(cov.obj_ref._values[property_name])
			
			# bundle it into a sequence
			property_value = PropertyValue(propertyIdentifier=property_name,value=Any(bundle_value),)
	
			# add it to the list
			list_of_values.append(property_value)
	
		# calculate time remaining
		if not cov.lifetime:
			time_remaining = 0
		else:
			time_remaining = int(cov.taskTime - current_time)
	
		# make sure it is at least one second
		if not time_remaining:
			time_remaining = 1
	
		# build a request with the correct type
		if cov.confirmed:
			request = ConfirmedCOVNotificationRequest()
		else:
			request = UnconfirmedCOVNotificationRequest()
	
		# fill in the parameters
		request.pduDestination = cov.client_addr
		request.subscriberProcessIdentifier = cov.proc_id
		request.initiatingDeviceIdentifier = cov.obj_ref._app.localDevice.objectIdentifier
		request.monitoredObjectIdentifier = cov.obj_id
		request.timeRemaining = time_remaining
		request.listOfValues = list_of_values
		# let the application send it
		cov.obj_ref._app.cov_notification(cov, request)