JoelBender/bacpypes #196: bacpypes/py27/bacpypes/bvllservice.py assumes an instance of the Address() Class to always have a .addrTuple property

Hoarfrost, 2018-06-10:
In **bacpypes/py27/bacpypes/bsllservice.py** the following lines
76    `# promote the normal and broadcast tuples`
77   ` self.addrTuple = self.address.addrTuple`
78    `self.addrBroadcastTuple = self.address.addrBroadcastTuple`

 expect the following properties of the self.address object to exist:
- self.address.addrTuple
- self.address.addrBroadcastTuple

There is no error checking to handle cases where these properties do not exist.

**self.address** is an instance of **Address()** defined in **bacpypes/py27/bacpypes/pdu.py**

If one creates an instance of Address initialized with an integer, the Address class never defines a property for either **.addrTuple** or **.addrBroadcastTuple**.
When this happens **bvllservice** encounters an unhandled exception:

This issue can be duplicated by running: `python IP2VLANRouger.py python 1 2 3`

`Traceback (most recent call last):
File "IP2VLANRouter.py", line 257, in <module> 
    main()
File "IP2VLANRouter.py", line 205, in main
    router = VLANRouter(local_address, local_network)
File "IP2VLANRouter.py", line 158, in __init__
    self.mux = UDPMultiplexer(local_address)
File "/Library/Python/2.7/site-packages/bacpypes-0.17.1-py2.7.egg/bacpypes/bvllservice.py", line 77, in __init__
    self.addrTuple = self.address.addrTuple
AttributeError: Address instance has no attribute 'addrTuple'`

Perhaps an alternative solution is to have Address always create these attributes.

JoelBender, 2018-06-11:
The first argument to the application must be the CIDR notation of the workstation/device that is running the application, for example, `192.168.0.15/24`.  This will be interpreted as an IPv4 address and it will have the appropriate tuples defined.  For addresses that are simple integers like what you provided, these are used for MS/TP and ARCNET networks and it doesn't make sense for them to have these tuples which are used in the socket API calls.

Hoarfrost, 2018-06-11:
After the fact I figured out that the first argument must be an IP address.  I didn't realize it should also have the subnet range specified as well until reading your comment.
However shouldn't the library still protect itself if someone uses the wrong form of the address and/or notify the user that a different type of address is expected.

JoelBender, 2018-06-11:
Yes, better error messages are better :-).  More seriously, there is a set of updates in another branch that will included in 0.18.0 where the application will be using a new `IPv4Address` class (or `IPv6Address`) class when it expects an IP address and the rejection will be a little clearer.  The sample applications could also fail "sooner" by making sure the address is correct, and the argument parser help strings could be better.

closed 2018-06-11 by JoelBender