JoelBender/bacpypes #199: W

spadilam, 2018-06-14:
Team

We are trying use readmutliple service where we are trying pass standard and proprietary properties.
We are facing getting the below error. Can you suggests the changes we need to make with respect to below code.

712
 ! <bacpypes.basetypes.ErrorType object at 0x026C1590>

systemStatus
 ! <bacpypes.basetypes.ErrorType object at 0x026C1550>

below is the code

    def do_read(self, args):
        """read <addr> ( <type> <inst> ( <prop> [ <indx> ] )... )..."""
        args = args.split()

        try:
            i = 0
            addr = args[i]
            i += 1

            read_access_spec_list = []
            while i < len(args):
                obj_type = args[i]
                i += 1

                if obj_type.isdigit():
                    obj_type = int(obj_type)
                elif not get_object_class(obj_type, ProprietaryObject.vendor_id):
                    if not get_object_class(obj_type):
                        raise ValueError("unknown object type")

                obj_inst = int(args[i])
                i += 1

                prop_reference_list = []
                while i < len(args):
                    prop_id = args[i]
                    if prop_id not in PropertyIdentifier.enumerations:
                        if prop_id.isdigit():
                            prop_id = int(prop_id)
                        # break

                    i += 1
                    if prop_id in ('all', 'required', 'optional'):
                        pass
                    else:
                        datatype = get_datatype(obj_type, prop_id)
                        if not datatype:
                            datatype = get_datatype(obj_type, prop_id, ProprietaryObject.vendor_id)
                            if not datatype:
                                raise ValueError("invalid property for object type")

                    # build a property reference
                    prop_reference = PropertyReference(
                        propertyIdentifier=prop_id,
                        )

                    # check for an array index
                    if (i < len(args)) and args[i].isdigit():
                        prop_reference.propertyArrayIndex = int(args[i])
                        i += 1

                    # add it to the list
                    prop_reference_list.append(prop_reference)

                # check for at least one property
                if not prop_reference_list:
                    raise ValueError("provide at least one property")

                # build a read access specification
                read_access_spec = ReadAccessSpecification(
                    objectIdentifier=(obj_type, obj_inst),
                    listOfPropertyReferences=prop_reference_list,
                    )

                # add it to the list
                read_access_spec_list.append(read_access_spec)

            # check for at least one
            if not read_access_spec_list:
                raise RuntimeError("at least one read access specification required")

            # build the request
            request = ReadPropertyMultipleRequest(
                listOfReadAccessSpecs=read_access_spec_list,
                )
            request.pduDestination = Address(addr)

            # make an IOCB
            iocb = IOCB(request)

            # give it to the application
            self.this_application.request_io(iocb)

            # wait for it to complete
            iocb.wait()

            # do something for success
            if iocb.ioResponse:
                apdu = iocb.ioResponse

                # should be an ack
                if not isinstance(apdu, ReadPropertyMultipleACK):
                    return

                # loop through the results
                for result in apdu.listOfReadAccessResults:
                    # here is the object identifier
                    objectIdentifier = result.objectIdentifier

                    # now come the property values per object
                    for element in result.listOfResults:
                        # get the property and array index
                        propertyIdentifier = element.propertyIdentifier
                        propertyArrayIndex = element.propertyArrayIndex

                        # here is the read result
                        readResult = element.readResult

                        print(propertyIdentifier)
                        if propertyArrayIndex is not None:
                            print("[" + str(propertyArrayIndex) + "]")

                        # check for an error
                        if readResult.propertyAccessError is not None:
                            print(" ! " + str(readResult.propertyAccessError) + '\n')

                        else:
                            # here is the value
                            propertyValue = readResult.propertyValue

                            # find the datatype
                            datatype = get_datatype(objectIdentifier[0], propertyIdentifier)
                            if not datatype:
                                raise TypeError("unknown datatype")

                            # special case for array parts, others are managed by cast_out
                            if issubclass(datatype, Array) and (propertyArrayIndex is not None):
                                if propertyArrayIndex == 0:
                                    value = propertyValue.cast_out(Unsigned)
                                else:
                                    value = propertyValue.cast_out(datatype.subtype)
                            else:
                                value = propertyValue.cast_out(datatype)

                            print(" = " + str(value) + '\n')
                        sys.stdout.flush()

            # do something for error/reject/abort
            if iocb.ioError:
                print(str(iocb.ioError) + '\n')

        except Exception as error:
            print("exception: %r", error)

JoelBender, 2018-06-14:
The `ErrorType` object has two components, `errorClass` and `errorCode`, check those and report back.

spadilam, 2018-06-15:
Below is the error we are getting:
BACNet - Connected
712
 ! <bacpypes.basetypes.ErrorType object at 0x0263B5B0>

Error Class: object
Error Code: unknownObject
systemStatus
 ! <bacpypes.basetypes.ErrorType object at 0x0263B570>

Error Class: object
Error Code: unknownObject

Error is getting in this piece of code:

                        # check for an error
                        if readResult.propertyAccessError is not None:
                            print (" ! " + str(readResult.propertyAccessError) + '\n')
                            print "Error Class:",readResult.propertyAccessError.errorClass
                            print "Error Code:",readResult.propertyAccessError.errorCode

JoelBender, 2018-06-15:
Note the value of the `objectIdentifier` from the `result.objectIdentifier` and that will tell you which object the device is complaining doesn't exist.

closed 2018-06-15 by JoelBender