JoelBender/bacpypes #200: Foreign device registration?

erincamille, 2018-06-15:
Hi,

I have a BACnet device that I would like to register as a foreign device on a BBMD. However, I can't seem to figure out how to get the device to initiate this request. Are there any examples of this? I see that there is a key for a foreignBBMD in the .ini file, but I don't see it used anywhere in the code. 

Thanks for your help!

JoelBender, 2018-06-15:
Look at the two sample applications **WhoIsIAm.py** and **WhoIsIAmForeign.py** and notice that the only differences are the application class, and that the version that registers itself as a foreign device has the extra parameters.  These are passed from the INI file down through the initialization code to the BVLL layer, compare the `BIPSimpleApplication` with the `BIPForeignApplication`.

erincamille, 2018-06-16:
Ok, thanks!

closed 2018-06-16 by erincamille