JoelBender/bacpypes #201: BACpypes COV not delivering values in samples

jklarson, 2018-06-15:
I am experimenting with the BACpypes COV samples in effort to understand how COV will be implemented in my project.

I have the server running with a binary value task, which reports updating the value locally.
I request a subscription to the COV object from the client. I receive a simple acknowledgement of the request from the server.

At this point, if the subscription was done correctly, I would expect to be receiving values from the server, which I am not. I suspect I am not subscribing properly. I am dubious as to what the first parameter of the subscribe command is actually asking for... clarification here may assist in solving the problem. 

Any help I could get in getting these examples running would be fantastic. Thanks

JoelBender, 2018-06-16:
The first parameter of the subscription, the `subscriberProcessIdentifier` is an "arbitrary" unsigned integer presented to the server that is stored with the subscription information and sent back with every notification, see clause _13.16.1.1.1_.  If your subscription request was acknowledged then in theory you subscribed correctly and if your not getting notifications then maybe the way you are changing the present value of the object isn't triggering it.  On the server add a "monitor" function to the object for the changes in present value:
```
    def ding(*args):
        print("ding! {}".format(args))

    some_object._property_monitors['presentValue'].append(ding)
```
This function should go "ding" whenever the value changes.

More likely you've come across a bug, the COV code has had only "light weight" testing.

JoelBender, 2018-07-01:
I've been working on the COV code, please check out the `192-cov-server-fails` branch for #192 and see if that solves this problem as well.

JoelBender, 2018-07-04:
Issue #192 is closed and is being released, so I'm closing this as well.

closed 2018-07-04 by JoelBender