JoelBender/bacpypes #202: Scripts for Reinitialize service & confirmed private transfer service

dayanand4u, 2018-06-18:
Hi,
    I am using few bacnet services(Read,write,Read multiple) provided in sample scripts for my work.Currently i am looking for few sample scripts to perform bacnet servcies like Reinitialize service & confirmed private transfer service .Could you please guide me about these servcies.

JoelBender, 2018-06-19:
Building an APDU for `ReinitializeDeviceRequest` on the client side is about as simple as it gets, it only has one required parameter `reinitializedStateOfDevice` that is an enumeration:
```
    request = ReinitializeDeviceRequest(reinitializedStateOfDevice='warmstart')
```
If you want to support these services in your BACpypes application, define a `do_ReinitializeDeviceRequest(self, apdu)` function which will receive the request, send back an ack, and then comes the harder part.

You only need to support COLDSTART and WARMSTART.  For a cold start I've had some limited success spawning a subprocess `subprocess.run(['/sbin/reboot'])` but I try to stay away from it.  A warm start is easier because you can wrap the executable in a shell script and check for a special return code from your application, like `sys.exit(5)` which runs your application again.

I've also had good luck with a "supervisor" application that runs other programs on a scheduled basis like cron and restarts those that fail.  Almost everything that application can do is now available with `systemd` so I've been moving (slowly) in that direction.

dayanand4u, 2018-06-20:
Thanks for the reply.I have tried to make Reinitialize service request in my application and its working fine.

closed 2018-06-20 by JoelBender