JoelBender/bacpypes #204: InvalidParameterDatatype thrown when writing int to presentValue

voney, 2018-06-21:
I'm getting an issue when using `WriteProperty()` to write an int to `presentValue` on an `analogOutputObject`.

Here's a snippet of what I'm doing, obj is an instance of `analogOutputObject`:

`key = 'presentValue'
value = 99
prop = prop = obj._attr_to_property(key)
prop.WriteProperty(obj, value)
`
This throws an `InvalidParameterDatatype ` exception with the message `presentValue must be of type Real`

I've tried casting `value` to `Real` using the below code but am getting the same issue:
`key = 'presentValue'
value = 99
prop = prop = obj._attr_to_property(key)
if isinstance(value, int):
    prop.WriteProperty(obj, Real(value))
else:
    prop.WriteProperty(obj, value)
`

If I write a float instead it works as expected.

Any pointers on what I may be doing wrong?

JoelBender, 2018-06-21:
Yep, try `prop.WriteProperty(obj, float(value))` and it should be fine.  The `presentValue` of an `AnalogOutputObject` is a `Real` which maps to a Python float.  The primitive value types have an `is_valid()` function so you don't end up writing -3 to an `Unsigned` and that kind of thing.

Since `Real(12)` is valid then an argument could be made that `int` should be valid for the `is_valid()` method, but I thought it was more important to be a little more restrictive (cue [robustness principle](https://en.wikipedia.org/wiki/Robustness_principle) discussion).

voney, 2018-06-22:
Thanks Joel, that's worked as expected. I appreciate the help. :)

closed 2018-07-01 by JoelBender