JoelBender/bacpypes #205: Weekly_Schedule Writin

clafouti2104, 2018-06-21:
Hi everybody,

I'm working on a schedule object. I can read it without any problem but I don't know how to write the weeklySchedule property:
I use the WritePropertyRequest console script, here's my command:

```
write 192.168.10.50 schedule 1 weeklySchedule [{(08:00,1),(17:00,0)}, {(08:00,1)),((08:00,1),(17:00,0)), ((08:00,1),(17:00,0),(19:00,1), (23:30,0)), ((08:00,1),(17:00,0)), ((00:00,0)),((10:00,1),(17:00,0))]
ERROR:__main__.ReadWritePropertyConsoleCmd:exception: ValueError("invalid literal for int() with base 10: '{(08:00,1)),((08:00,1),(17:00,0)),'",)
Traceback (most recent call last):
  File "write.py", line 139, in do_write
    indx = int(args[5])
```
or 
```
> write 192.168.10.50 schedule 1 weeklySchedule [     [{"01:00:00",10}],     [{"01:00:00", 10},{"02:00:00",20}],[{"01:00:00", 10}, {"02:00:00", 20},{"03:00:00", 30}],[],[], [],[] ]
ERROR:__main__.ReadWritePropertyConsoleCmd:exception: ValueError('invalid literal for int() with base 10: \'[{"01:00:00",10}],\'',)
Traceback (most recent call last):
  File "write.py", line 139, in do_write
    indx = int(args[5])
```

Can you help me please?

JoelBender, 2018-06-21:
The console application you are using isn't sophisticated enough to parse a weekly schedule.  Looking at your sample command I can understand what you are trying to write, so it wouldn't be too bad writing a simple parser for it.  Look in the **LocalScheduleObject.py** sample code for the kind of data structure you need to build, warm up your fingers for some regex, [parsy](https://github.com/python-parsy/parsy), or [pyparsing](https://github.com/pyparsing/pyparsing) code, or if you're really not picky, `eval()`.

JoelBender, 2019-11-17:
Closing this as resolved, please upgrade to **0.17.7** and if you still need help with it please create a new issue.

closed 2019-11-17 by JoelBender