JoelBender/bacpypes #206: Receive response but not catched by BACPypes

Enderdead, 2018-06-27:
Hello,
I  just getting started to test some sample code in order to get data from a device working with BACnet/IP.
I using bacpypes-0.17.2.
I tried many sample script like : WhoIAm.py and ReadProperty.py.
In every command (whois, read), i can see with WireShark my request and the response from the target  but BACPypes  didn"t see them.
My IP is : 192.168.2.26 
The Target is : 192.168.2.15
The netmask is : 255.255.255.0

This is my my BACpypes.ini :
```
[BACpypes]
objectName: Betelgeuse
address: 192.168.2.0/24
objectIdentifier: 2000
maxApduLengthAccepted: 1476
segmentationSupported: segmentedBoth
vendorIdentifier: 15
foreignBBMD: 192.168.2.15
foreignTTL: 30

```
In order to illustrate my problem, when I launch this command
 read 192.168.2.15 analogInput 7 presentValue

I can this that in Wireshare

[image]

[image]

Has anyone  can provide some solution about my problem?

Thank's

dumbledore-lessorciersweb, 2018-06-29:
Hi
I think you have to put on the BACpypes.ini the address of the device and not the network
In your case the address should be 192.168.2.26/24 and not 192.168.2.0/24.

The Foreign address shouldn't be the address of the destination but I don't think this can make the BACpypes not communicate in your case.

Enderdead, 2018-07-05:
Thank's a lot ! Everything works perfectly.

closed 2018-07-05 by Enderdead