JoelBender/bacpypes #207: Trend Log logBuffer explanation

dumbledore-lessorciersweb, 2018-06-30:
Hi
I tried today to implement some trend logs on my device and I didnt know how to put some datas on logBuffer.
I tried to put some fixed data's like this to test.
LogBuffer= SequenceOf(LogRecord)([LogRecord(
Timestamp = Datetime(...), LogDatum=(realValue=Real(7.0)),...
])
When I open the TrendLog with another BACnet explorer i didn't see any logs in...
Could you please try to put me in the right direction?

Thanks.

dumbledore-lessorciersweb, 2018-06-30:
Thanks I will try this!
Have a nice weekend

JoelBender, 2018-06-30:
I had a chance to mock up an in-memory `TrendLog` that might be useful in a [gist](https://gist.github.com/JoelBender/49dd7f3100fdbb1886bb463f52ceb7db), and shows the direction that I am going.  The next step would be to put this in a `bacpypes.local.trendlog` module along with a `LocalTrendLogObject` implementation, and then put `ReadRangeServices` in `bacpypes.service.object`.

epaulson, 2018-08-12:
I'm pretty well confused on how to implement the ReadRangeService - I think specifically on how to convert the log buffer into a ReadRangeACK itemData structure.    

I'm testing with the ReadRange.py sample code. I spun up a simple server, stole some code from services from readPropertyRequest and mixed it into my app, and tried to trim down to what I think is a minimal ReadRangeRequest handler, like so:

```python
    def do_ReadRangeRequest(self, apdu):
        """Return the value of some property of one of our objects."""
        print("Servicing a readrange request of %s" % (apdu))
        if _debug: ReadRangeServices._debug("do_ReadPropertyRequest %r", apdu)
        resp = ReadRangeACK(context=apdu)

        data_log = []
        datum = LogRecordLogDatum()
        setattr(datum, 'realValue', 10.0)

        log_record = LogRecord(
            timestamp=DateTime(date=Date().now().value, time=Time().now().value),
            logDatum=datum,
            )
        data_log.append(log_record)
        any_datalog = []
        for a in data_log:
          temp = Any()
          temp.cast_in(a)
          any_datalog.append(temp)

        #resp.itemData = Any()
        #resp.itemData.cast_in(SequenceOf(Any)(any_datalog))
        resp.itemData = SequenceOf(Any)(any_datalog)
        resp.objectIdentifier = apdu.objectIdentifier
        resp.propertyIdentifier = apdu.propertyIdentifier
        resp.propertyArrayIndex = apdu.propertyArrayIndex
        resp.resultFlags = ResultFlags([0,1,0])
        resp.itemCount = 1

        resp.firstSequenceNumber = 0
        print(resp)
        self.response(resp)
```
This code blows up with an "invalid constructor error" - here's debugging turned on rather high:

```
DEBUG:bacpypes.constructeddata.Sequence:__init__ () {}
DEBUG:bacpypes.constructeddata.Sequence:    - my_kwargs: {}
DEBUG:bacpypes.constructeddata.Sequence:    - other_kwargs: {}
DEBUG:bacpypes.constructeddata.Sequence:decode <bacpypes.primitivedata.TagList object at 0x7fc677694be0>
Servicing a readrange request of <bacpypes.apdu.ReadRangeRequest(26,35) instance at 0x7fc677694b00>
DEBUG:bacpypes.constructeddata.Sequence:__init__ () {}
DEBUG:bacpypes.constructeddata.Sequence:    - my_kwargs: {}
DEBUG:bacpypes.constructeddata.Sequence:    - other_kwargs: {}
DEBUG:bacpypes.constructeddata.Choice:__init__ {}
DEBUG:bacpypes.constructeddata.Choice:    - my_kwargs: {}
DEBUG:bacpypes.constructeddata.Choice:    - other_kwargs: {}
DEBUG:bacpypes.constructeddata.Sequence:__init__ () {'date': (118, 8, 12, 7), 'time': (2, 20, 0, 13)}
DEBUG:bacpypes.constructeddata.Sequence:    - my_kwargs: {'date': (118, 8, 12, 7), 'time': (2, 20, 0, 13)}
DEBUG:bacpypes.constructeddata.Sequence:    - other_kwargs: {}
DEBUG:bacpypes.constructeddata.Sequence:__init__ () {'timestamp': <bacpypes.basetypes.DateTime object at 0x7fc677694c50>, 'logDatum': <bacpypes.basetypes.LogRecordLogDatum object at 0x7fc677694d30>}
DEBUG:bacpypes.constructeddata.Sequence:    - my_kwargs: {'timestamp': <bacpypes.basetypes.DateTime object at 0x7fc677694c50>, 'logDatum': <bacpypes.basetypes.LogRecordLogDatum object at 0x7fc677694d30>}
DEBUG:bacpypes.constructeddata.Sequence:    - other_kwargs: {}
DEBUG:bacpypes.constructeddata.Any:cast_in <bacpypes.basetypes.LogRecord object at 0x7fc677694cf8>
DEBUG:bacpypes.constructeddata.Sequence:encode <bacpypes.primitivedata.TagList object at 0x7fc677694ef0>
DEBUG:bacpypes.constructeddata.Sequence:encode <bacpypes.primitivedata.TagList object at 0x7fc677694ef0>
DEBUG:bacpypes.constructeddata.Sequence:    - build helper: <class 'bacpypes.primitivedata.Date'> (118, 8, 12, 7)
DEBUG:bacpypes.constructeddata.Sequence:    - build helper: <class 'bacpypes.primitivedata.Time'> (2, 20, 0, 13)
DEBUG:bacpypes.constructeddata.Choice:('LogRecordLogDatum')encode <bacpypes.primitivedata.TagList object at 0x7fc677694ef0>
DEBUG:bacpypes.constructeddata.SequenceOf:SequenceOf <class 'bacpypes.constructeddata.Any'>
DEBUG:bacpypes.constructeddata.SequenceOf:SequenceOf <class 'bacpypes.constructeddata.Any'>
DEBUG:bacpypes.constructeddata.SequenceOf:    - found in cache
DEBUG:bacpypes.constructeddata.SequenceOf:    - found in cache
DEBUG:bacpypes.constructeddata._SequenceOf:('SequenceOfAny')__init__ [<bacpypes.constructeddata.Any object at 0x7fc677694e10>] (subtype=<class 'bacpypes.constructeddata.Any'>)
[... 11 lines cut ...]
    helperFn(apdu)
  File "ReadRangeServer.py", line 95, in do_ReadRangeRequest
    self.response(resp)
  File "/usr/local/lib/python3.6/site-packages/bacpypes/comm.py", line 482, in response
    self.elementService.sap_confirmation(*args,**kwargs)
  File "/usr/local/lib/python3.6/site-packages/bacpypes/appservice.py", line 1501, in sap_confirmation
    apdu.encode(xpdu)
  File "/usr/local/lib/python3.6/site-packages/bacpypes/apdu.py", line 672, in encode
    Sequence.encode(self, self._tag_list)
  File "/usr/local/lib/python3.6/site-packages/bacpypes/constructeddata.py", line 99, in encode
    helper = element.klass(value)
  File "/usr/local/lib/python3.6/site-packages/bacpypes/constructeddata.py", line 392, in __init__
    raise TypeError("invalid constructor datatype")
TypeError: invalid constructor datatype
DEBUG:bacpypes.constructeddata.Sequence:__init__ () {'errorClass': 'device', 'errorCode': 'operationalProblem'}
DEBUG:bacpypes.constructeddata.Sequence:    - my_kwargs: {'errorClass': 'device', 'errorCode': 'operationalProblem'}
DEBUG:bacpypes.constructeddata.Sequence:    - other_kwargs: {}
[... cut ...]
DEBUG:bacpypes.constructeddata.Sequence:    - build helper: <class 'bacpypes.basetypes.ErrorCode'> 'operationalProblem'
```

I've tried a bunch of variations fiddling with what the SequenceOf(type) should be, and if the payload should be converted to Any() or just left alone, but I think I'm just lost with when do to the casting.

This is with a 0.16 bacpypes, if it matters (I think 16.4) - if upgrading would help or make it easier to debug I'm happy to do so, I just have a working docker image that I'm avoiding upgrading unless I have to.

JoelBender, 2018-08-12:
Yep, this is not very clear.  First, building a `LogRecord` should be somewhat familiar:
```
>>> from bacpypes.primitivedata import Date, Time
>>> from bacpypes.basetypes import DateTime, LogRecord, LogRecordLogDatum

>>> lrld = LogRecordLogDatum(realValue=75.3)
>>> dt = DateTime(date=Date().now().value, time=Time().now().value)
>>> lr1 = LogRecord(timestamp=dt, logDatum=lrld)
>>> lr1.debug_contents()
    timestamp
        date = (118, 8, 11, 6)
        time = (23, 12, 37, 43)
    logDatum
        realValue = 75.3
```
The `Any` can take any contsructed data type, but this is a specific data type which we need to build first:
```
>>> from bacpypes.constructeddata import SequenceOf, Any
>>> 
>>> SequenceOfLogRecord = SequenceOf(LogRecord)
>>> solr = SequenceOfLogRecord([lr1])
```
So now `solr` is an instance of a sequence of log records.  Now build the `ReadRangeACK` and put in a placeholder `Any` where the value is going to go, then cast-in the sequence.
```
>>> from bacpypes.apdu import ReadRangeACK
>>> rra = ReadRangeACK(
    objectIdentifier=('analogValue', 1),
    propertyIdentifier='presentValue',
    resultFlags=[1, 0, 1],
    itemCount=1,
    itemData=Any(),
    )
>>> rra.itemData.cast_in(solr)
```
The other end of the service is going to need the same data type to be able to cast it out:
```
>>> z = rrs.itemData.cast_out(SequenceOfLogRecord)
```
The result is _not_ an instance like it was going it (which I guess is a bit
unfortunate) but just a Python list:
```
>>> z
[<bacpypes.basetypes.LogRecord object at 0x7fc7f2ac7850>]
>>> 
>>> z[0].debug_contents()
    timestamp
        date = (118, 8, 11, 6)
        time = (23, 12, 37, 43)
    logDatum
        realValue = 75.30000305175781
```
Hopefully that helps.

epaulson, 2018-08-12:
Thanks, it is helpful for understanding. However, I'm still having some trouble - I'm not sure if this is supposed to work or not, but I get:

```
>>> rra.debug_contents()
    pduExpectingReply = 0
    pduNetworkPriority = 0
    apduType = 3
    apduService = 26
    objectIdentifier = ('analogValue', 1)
    propertyIdentifier = 'presentValue'
    resultFlags = [1, 0, 1]
    itemCount = 1
    itemData
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/vagrant/miniconda2/envs/bacnet/lib/python3.6/site-packages/bacpypes/debugging.py", line 214, in debug_contents
    klass.debug_contents(self, indent, file, _ids)
  File "/home/vagrant/miniconda2/envs/bacnet/lib/python3.6/site-packages/bacpypes/constructeddata.py", line 326, in debug_contents
    helper = element.klass(value)
  File "/home/vagrant/miniconda2/envs/bacnet/lib/python3.6/site-packages/bacpypes/constructeddata.py", line 419, in __init__
    raise TypeError("invalid constructor datatype")
TypeError: invalid constructor datatype
```
This is with bacpypes 0.17.4 and python 3.6.6.  

This is similar to the error I see when I call .respond() at the end of my ReadRange handler, but I might not be doing that correctly. If there's a better way to simulate the marshaling of data before it's sent back to the requester, I'm happy to try that instead.

JoelBender, 2018-08-12:
Ah!  This is a sticky wicket.  Until I can think more about how to make `SequenceOf(LogRecord)` a subtype of `SequenceOf(Any)` (probably in a similar direction to `AnyAtomic`) then change...
```
Element('itemData', SequenceOf(Any), 5)
```
to
```
Element('itemData', Any, 5)
```
In the `bacpypes.apdu` module.  This might have to be another weird corner case where there is no really clean answer.  After you cast the value "in" the debugging will just show you the tag list because all sense of structure has been lost and it's not until it is cast "out" that the structure returns.

I'm marking this as a bug for the next release after 0.17.5.

epaulson, 2018-08-12:
Thanks. I made that change to my local install of Bacpypes and now ReadRange.py successfully reads from my test TrendLog server, as does YABE.

JoelBender, 2019-04-19:
Please check out the `207-trend-logbuffer` branch and see if there's anything you need to change in your code.  I added a `SequenceOfAny` class with `cats_in()` and `cast_out()` functions and there is sample code for a client issuing a Read Range Request and a really, really (no, really) lame server sample with a single log record.

The next step is to create a "LocalTrendLogObject" with all the right stuff for maintaining the log which will need an additional API for interpreting the `Range` choices as well as something like `log_status()`, `log_value()`, `log_failure()`, `log_time_change()`.

JoelBender, 2019-04-19:
I decided to release these changes now and I'll make a new enhancement request for the next step.

closed 2019-04-19 by JoelBender