JoelBender/bacpypes #216: Waiting For Link Status

JoelBender, 2018-08-15:
In [this discussion](https://gitter.im/JoelBender/bacpypes?at=5b746a0f637ee66082f53250) there needs to be a way for applications to know if the link (all links? some links?) are ready for downstream packets.  This could be waiting for foreign device registration with IPv4/IPv6, connection ready for SC, connections and subscriptions acked for MQTT, exchanges and queues ready for AMPQ, etc.

mdsteveb, 2018-08-20:
I don't know if you've actually tested this yet but when I run this after customizing the points I'm looking for, I get this:

```
$ ./MultipleReadPropertyForeign.py --debug __main__
DEBUG:__main__:initialization
DEBUG:__main__:    - args: Namespace(buggers=False, color=False, debug=['__main__'], ini=<class 'bacpypes.consolelogging.ini'>)
DEBUG:__main__:    - this_device: <bacpypes.local.device.LocalDeviceObject object at 0x10360c470>
DEBUG:__main__.ReadPointListApplication:__init__ [('31:10', 'analogInput:7', 'presentValue'), ('31:10', 'analogInput:8', 'presentValue')], (<bacpypes.local.device.LocalDeviceObject object at 0x10360c470>, '192.168.160.232', <Address 192.168.172.103>, 30)
DEBUG:__main__.PrairieDog:__init__ 500
DEBUG:__main__:    - dog: <__main__.PrairieDog object at 0x10302a080>
    <__main__.PrairieDog object at 0x10302a080>
        isScheduled = False
        taskInterval = 500
DEBUG:__main__:running
DEBUG:__main__.PrairieDog:process_task
DEBUG:__main__.PrairieDog:    - success
DEBUG:__main__.ReadPointListApplication:next_request
DEBUG:__main__.PrairieDog:process_task
DEBUG:__main__.PrairieDog:    - success
DEBUG:__main__.ReadPointListApplication:next_request
DEBUG:__main__.PrairieDog:process_task
DEBUG:__main__.PrairieDog:    - success
DEBUG:__main__.ReadPointListApplication:next_request
DEBUG:__main__.ReadPointListApplication:    - done
DEBUG:__main__:fini
```

...with no result. Should be able to dig more later tonight.

JoelBender, 2019-04-15:
postponed for a later release