JoelBender/bacpypes #217: BBMD not passing traffic from host network using virtual boxes

JefferyMarshall, 2018-08-29:
We are using your BBMD software to simulate bridging 2 subnets. 
We have one BBMD listing to the broadcast on the subnet and sending to the other BBMD on the opposing subnet. So our when running the script we pass in the following arguments.

subnet0:
BBMD.py 192.168.56.255 192.168.57.6

At this point using wireshark we see the broadcast on the network from the simulator. Then we see the broadcast received by the BBMD in the logs, however we do not see the data get passed to the other subnet. We do not see the BBMD do any sending of any kind.

Our set up has a couple of virtual machines on the same network, each running a simulator and a BBMD. The other subnet has the same. Each BBMD virtual machine is running and looking at the other BBMD on the other network.

Why is it that the traffic from the host subnet being received but not being passed to the second subnet?

JoelBender, 2018-08-30:
It looks like the first parameter is a broadcast address and it needs to be the device address and subnet, like `192.168.56.7/24`.  By saying you can see the PDU coming into the BBMD in the logs, is that from running with `--debug bacpypes.bvllservice.BIPBBMP` and seeing the "confirmation" function being called with the OriginalBroadcastNPDU?  Try also running it with `--debug bacpypes.core` and see if an exception is being thrown.

JefferyMarshall, 2018-08-31:
Sorry, I misunderstood how to set the first parameter, I tried a few different combinations and listening on the broadcast was the only one that showed any results.
After setting the first devices address and the subnet worked great. 
Thanks.

JefferyMarshall, 2018-08-31:
I am seeing a response on the second simulator in the logs. I don't see any traffic from wireshark on the second network and I don't see a the iam messages coming from the second network to the first.
Here are the arguments that I am passing

Simulator 1
python BBMD.py 192.168.56.104/24 192.168.57.7 --debug bacpypes.core
response : 192.168.56.255:47808
Simulator 2
python BBMD.py 192.168.57.7 192.168.56.104 --debug bacpypes.core
response : 192.168.57.7:47808

I have a feeling that there is just a problem with the arguments. does what I have look right to you?

JoelBender, 2019-11-17:
Closing this as resolved.

closed 2019-11-17 by JoelBender