JoelBender/bacpypes #218: help building simple interface for one hvac unit

martialarturo, 2018-09-11:
Hello,

Python/Bacnet newbie here.

In short, I'm trying to build a simple interface that will display (and refresh periodically) the current temperatures in 4 rooms. Eventually I'd like to incorporate adjusting these in the same application.

I have figured out how to read the temps via ReadProperty.py but haven't figured out how to integrate an action similar to "read analogValue 5 presentValue" to display a room temp.

Can/will anyone point me in the right direction? Thanks!

JoelBender, 2018-09-12:
What kind of display would you like to see?  There are a bunch of desktop application frameworks [like these](https://docs.python-guide.org/scenarios/gui/), I don't have experience other than "run the demo" with any one of them.  For auto-refresh web pages there's everything from a "polling loop" in Javascript that uses a web service interface and a little bit of jQuery to full blown Angular/React/Vue single page application.

JoelBender, 2018-09-13:
I made a [sample application](https://gist.github.com/JoelBender/d6c55d934d285869f4bab364ab040a02) using [Flask](http://flask.pocoo.org/docs/1.0/) that you might find interesting.  There is a BACpypes warning about signal handlers, and there's some weirdness with the debugging mode for Flask, and it's not for a production environment.  When BACpypes jumps into the Python 3.5+ async/await world, all the thread problems will go away and we can use [Sanic](https://sanic.readthedocs.io/en/latest/) or something like it.

martialarturo, 2018-09-14:
I really appreciate your helpfulness to a noob such as myself. I started this intending for it to be a learning project, but this area seems less common than other development projects and therefore is more difficult to research.

When I try to run FlaskReadProperty.py I get the following error:
```
Traceback (most recent call last):
  File "FlaskReadProperty.py", line 16, in <module>
    from bacpypes.iocb import IOCB
ImportError: No module named iocb
```
Am I missing a file somewhere, missing a library (like I have installed an incorrect version of bacpypes), or am I missing something else? The more I grow as a developer (very novice), the more I realize I don't know.

Thanks Again,
Art

martialarturo, 2018-09-19:
Thanks a ton! This put me well down the path!!