JoelBender/bacpypes #220: exception in whois handling in 0.17.5

epaulson, 2018-09-16:
I've got a test program that's basically a modified version of the Random Value Property sample, though I added on some COV handling, and a background thread that's reading from a data source so I can make it available as a BACnet point. 

It's using a version of bacpypes checked out from Github rather than pypi, so I can incorporate the workaround in 207 for a future test program. Git tells me the checkout is 61dbd5237fe4406f64755c8cd7526ce28f5eb809 which is "0.17.5 release"

All seemed fine for about two weeks, but recently it seems to continuously have problems with a whois request, presumably coming from someone else poking around the bacnet network:

```python
ERROR:bacpypes.app.Application:exception: KeyError(None,)
Traceback (most recent call last):
  File "/home/pi/bacnet/lib/python3.5/site-packages/bacpypes-0.17.5-py3.5.egg/bacpypes/app.py", line 372, in indication
    helperFn(apdu)
  File "/home/pi/bacnet/lib/python3.5/site-packages/bacpypes-0.17.5-py3.5.egg/bacpypes/service/device.py", line 102, in do_WhoIsRequest
    self.i_am(address=apdu.pduSource)
  File "/home/pi/bacnet/lib/python3.5/site-packages/bacpypes-0.17.5-py3.5.egg/bacpypes/service/device.py", line 127, in i_am
    self.request(iAm)
  File "/home/pi/bacnet/lib/python3.5/site-packages/bacpypes-0.17.5-py3.5.egg/bacpypes/app.py", line 460, in request
    super(ApplicationIOController, self).request(apdu)
  File "/home/pi/bacnet/lib/python3.5/site-packages/bacpypes-0.17.5-py3.5.egg/bacpypes/app.py", line 354, in request
    super(Application, self).request(apdu)
  File "/home/pi/bacnet/lib/python3.5/site-packages/bacpypes-0.17.5-py3.5.egg/bacpypes/comm.py", line 574, in request
    self.elementService.sap_indication(*args, **kwargs)
  File "/home/pi/bacnet/lib/python3.5/site-packages/bacpypes-0.17.5-py3.5.egg/bacpypes/appservice.py", line 1501, in sap_indication
    self.request(xpdu)
  File "/home/pi/bacnet/lib/python3.5/site-packages/bacpypes-0.17.5-py3.5.egg/bacpypes/comm.py", line 574, in request
    self.elementService.sap_indication(*args, **kwargs)
  File "/home/pi/bacnet/lib/python3.5/site-packages/bacpypes-0.17.5-py3.5.egg/bacpypes/appservice.py", line 1322, in sap_indication
    self.request(apdu)
  File "/home/pi/bacnet/lib/python3.5/site-packages/bacpypes-0.17.5-py3.5.egg/bacpypes/comm.py", line 282, in request
    self.clientPeer.indication(*args, **kwargs)
  File "/home/pi/bacnet/lib/python3.5/site-packages/bacpypes-0.17.5-py3.5.egg/bacpypes/netservice.py", line 292, in indication
    adapter = self.local_adapter or self.adapters[None]
KeyError: None
```

I have a pretty minimal .ini file - we run on a non-standard port so that's a little different
```
[BACpypes]
objectname: testobject
objectIdentifier: 133799
address = x.x.y.y/24:47818
maxApduLengthAccepted: 1024
segmentationSupported: segmentedBoth
vendorIdentifier: 15
```

Is there something more I'm supposed to include in the ini file to handle this case? Do I need to implement some code to properly complete i_am handling?

This is kind of vague bug report so I'm happy to try to gather a better set of data on the issue, but I thought I would post quick in case it was something obvious.

JoelBender, 2018-09-17:
The `local_adapter` is set when there is more than one adapter bound to the network service access point and it should be the one considered "local" to the application layer above it.  A BACnet device should have one and only one BACnet address, which includes the network number, so if the device is also a router, only one of its directly connected networks is considered local.

Most "simple" applications have no idea what network they are on, they only know that there are "local" addresses and "remote" addresses.  In that case there is no `local_network` set, and `self.adapters[None]` references the one and only bound adapter.  There is new code for `NetworkNumberIs` and you'll notice that when the local network is learned that `self.adapters[None]` is deleted (but the object that it references doesn't go away, the `adapter` parameter should reference the same object.

So I'm wondering if at some point your network interfaces is receiving a `NetworkNumberIs` message, the network service element is "learning" the locally connected network, but by failing to set the `local_adapter` the other incoming messages are throwing exceptions because the `None` reference has gone away.

If you could, please confirm this hypothesis by seeing if you are receiving a `NetworkNumberIs` message just before the errors show up.

epaulson, 2018-09-17:
I have restarted my program with ```--debug bacpypes.netservice.NetworkServiceElement``` and will keep an eye on it, but it looks like whatever was sending me whois requests has stopped. I'll try to force some myself.

epaulson, 2018-10-21:
I flipped back to 0.17.4 for my app, and haven't had a chance to stick another box on the same network with 0.17.5 to see if I could reproduce it, but I did leave it with the same `--debug bacpypes.netservice.NetworkServiceElement` flag running. Here's that output - which again, because its running 0.17.4, I suppose doesn't necessarily show that it's the same activity that caused the crash. Sorry about the long delay here, I hope to get another network port on that network and to be able to bring up a second box to test with, but no go so far. 

Note that neither 10.a.b.c nor 10.x.y.z are on the same /24 as my box - indeed, 10.x.y.z is an entirely different /16 than my box.

```
DEBUG:bacpypes.netservice.NetworkServiceElement:indication <bacpypes.netservice.
NetworkAdapter object at 0x761fac90> <bacpypes.npdu.RejectMessageToNetwork objec
t at 0x761ae4b0>
    <bacpypes.netservice.NetworkAdapter object at 0x761fac90>
        adapterSAP = <bacpypes.netservice.NetworkServiceAccessPoint object at 0x
761fa9b0>
    <bacpypes.npdu.RejectMessageToNetwork object at 0x761ae4b0>
        pduSource = <Address 10.a.b.c:47818>
        pduDestination = <LocalBroadcast *>
        pduExpectingReply = False
        pduNetworkPriority = 1
        npduVersion = 1
        npduControl = 129
        npduNetMessage = 3
        rmtnDNET = 6500
        pduData = x''
DEBUG:bacpypes.netservice.NetworkServiceElement:RejectMessageToNetwork <bacpypes
.netservice.NetworkAdapter object at 0x761fac90> <bacpypes.npdu.RejectMessageToN
etwork object at 0x761ae4b0>
    <bacpypes.netservice.NetworkAdapter object at 0x761fac90>
        adapterSAP = <bacpypes.netservice.NetworkServiceAccessPoint object at 0x
761fa9b0>
    <bacpypes.npdu.RejectMessageToNetwork object at 0x761ae4b0>
        pduSource = <Address 10.a.b.c:47818>
        pduDestination = <LocalBroadcast *>
        pduExpectingReply = False
        pduNetworkPriority = 1
        npduVersion = 1
        npduControl = 129
        npduNetMessage = 3
        rmtnDNET = 6500
        pduData = x''
DEBUG:bacpypes.netservice.NetworkServiceElement:indication <bacpypes.netservice.
NetworkAdapter object at 0x761fac90> <bacpypes.npdu.IAmRouterToNetwork object at
 0x761aea10>
    <bacpypes.netservice.NetworkAdapter object at 0x761fac90>
        adapterSAP = <bacpypes.netservice.NetworkServiceAccessPoint object at 0x
761fa9b0>
    <bacpypes.npdu.IAmRouterToNetwork object at 0x761aea10>
        pduSource = <Address 10.x.y.z:47818>
        pduDestination = <LocalBroadcast *>
        pduExpectingReply = False
        pduNetworkPriority = 0
        npduVersion = 1
        npduControl = 128
        npduNetMessage = 1
        iartnNetworkList = [3]
        pduData = x''
DEBUG:bacpypes.netservice.NetworkServiceElement:IAmRouterToNetwork <bacpypes.net
service.NetworkAdapter object at 0x761fac90> <bacpypes.npdu.IAmRouterToNetwork o
bject at 0x761aea10>
    <bacpypes.netservice.NetworkAdapter object at 0x761fac90>
        adapterSAP = <bacpypes.netservice.NetworkServiceAccessPoint object at 0x
761fa9b0>
    <bacpypes.npdu.IAmRouterToNetwork object at 0x761aea10>
        pduSource = <Address 10.x.y.z:47818>
        pduDestination = <LocalBroadcast *>
        pduExpectingReply = False
        pduNetworkPriority = 0
        npduVersion = 1
        npduControl = 128
        npduNetMessage = 1
        iartnNetworkList = [3]
        pduData = x''
DEBUG:bacpypes.netservice.NetworkServiceElement:    - sap: <bacpypes.netservice.
[... 129 lines cut ...]
        pduSource = <Address 10.a.b.c:47818>
        pduDestination = <LocalBroadcast *>
        pduExpectingReply = False
        pduNetworkPriority = 1
        npduVersion = 1
        npduControl = 129
        npduNetMessage = 3
        rmtnDNET = 6500
        pduData = x''
DEBUG:bacpypes.netservice.NetworkServiceElement:RejectMessageToNetwork <bacpypes
.netservice.NetworkAdapter object at 0x761fac90> <bacpypes.npdu.RejectMessageToN
etwork object at 0x761c7450>
    <bacpypes.netservice.NetworkAdapter object at 0x761fac90>
        adapterSAP = <bacpypes.netservice.NetworkServiceAccessPoint object at 0x
761fa9b0>
    <bacpypes.npdu.RejectMessageToNetwork object at 0x761c7450>
        pduSource = <Address 10.a.b.c:47818>
        pduDestination = <LocalBroadcast *>
        pduExpectingReply = False
        pduNetworkPriority = 1
        npduVersion = 1
        npduControl = 129
        npduNetMessage = 3
        rmtnDNET = 6500
        pduData = x''

JoelBender, 2018-10-23:
Interesting pattern, but probably not enough to diagnose anything.  Basically it's 10.a.b.c:47818 telling everyone it is *not* a router to network 6500 and 10.x.y.z:47818 saying it *is* a router to network 3.  All of these are being received because they were locally broadcast.  More information gleefully accepted!

epaulson, 2018-12-26:
I had a chance on the Friday before the holiday to set up some boxes on the same network with 0.17.5 and let them run with debugging, and Santa brought me a crash. Sure enough, in the debugging output from bacpypes right before the crash, I see a NetworkNumberIs message. 

Do you mind if I email you the trace? That way I don't have to scrub out the IP addresses. I will use the address in the PyPI page for bacpypes.

JoelBender, 2018-12-27:
Ok, I really don't want to know what _else_ you have on your Christmas List! :smile:  Yes, please send me the debug trace, and if it's too big for email we'll arrange something else.

JoelBender, 2019-04-11:
@epaulson I'm getting ready to release 0.17.6, please restart your testing when you get the new release

epaulson, 2019-05-24:
So the good news is that the problem is easy to reproduce. The bad news is that it still shows up in 0.17.6. 

It turns out that I can just fire up the WhoIsIAm example and let it sit for a couple of days and it will eventually get hit. I ran a Pi with 0.17.5 and a 0.17.6, and both of them failed after a few days, probably at the same time (I checked once a day.) Unfortunately I didn't have a 3rd Pi handy to run 0.17.4 on to see if it kept going.

We have a very busy BACnet network with lots of devices (some of them weird) so it wouldn't be surprising to me that if there's a device that might send out non-conforming BACnet messages, there's a good chance we've got one.

Here's a failure. Is there a logging setting you'd like me to turn on and send you via email? 

```python
(bacpypes176) pi@raspberrypi:~/bacpypes/samples $ python WhoIsIAm.py --ini ../Base.ini
> exception: KeyError(None,)
Traceback (most recent call last):
  File "/home/pi/bacpypes176/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/app.py", line 372, in indication
    helperFn(apdu)
  File "/home/pi/bacpypes176/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/service/device.py", line 102, in do_WhoIsRequest
    self.i_am(address=apdu.pduSource)
  File "/home/pi/bacpypes176/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/service/device.py", line 127, in i_am
    self.request(iAm)
  File "WhoIsIAm.py", line 54, in request
    BIPSimpleApplication.request(self, apdu)
  File "/home/pi/bacpypes176/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/app.py", line 460, in request
    super(ApplicationIOController, self).request(apdu)
  File "/home/pi/bacpypes176/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/app.py", line 354, in request
    super(Application, self).request(apdu)
  File "/home/pi/bacpypes176/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/comm.py", line 574, in request
    self.elementService.sap_indication(*args, **kwargs)
  File "/home/pi/bacpypes176/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/appservice.py", line 1505, in sap_indication
    self.request(xpdu)
  File "/home/pi/bacpypes176/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/comm.py", line 574, in request
    self.elementService.sap_indication(*args, **kwargs)
  File "/home/pi/bacpypes176/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/appservice.py", line 1326, in sap_indication
    self.request(apdu)
  File "/home/pi/bacpypes176/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/comm.py", line 282, in request
    self.clientPeer.indication(*args, **kwargs)
  File "/home/pi/bacpypes176/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/netservice.py", line 292, in indication
    adapter = self.local_adapter or self.adapters[None]
KeyError: None
```

JoelBender, 2019-05-25:
If you put this line after the creation of the `WhoIsIAmApplication()` in the application (it should be about line 187):
```python
    this_application.nsap.debug_contents()
```
You should see some output like this:
```python
    adapters = {
        None : <bacpypes.netservice.NetworkAdapter object at 0x7f9f302a8ad0>
            adapterSAP = <bacpypes.netservice.NetworkServiceAccessPoint object at 0x7f9f302bb310>
        }
    pending_nets = {}
    local_adapter = <bacpypes.netservice.NetworkAdapter object at 0x7f9f302a8ad0>
    local_address = <Address 10.0.1.93>
```
This means that the Network Service Access Point has one "local" downstream path and that it is not bound to a specific network.  If there were more elements in the `adapters` dictionary then those would be network numbers and the network layer of the application would be acting like a router. The `local_adapter` references the one to be considered "local" for the application, and notice that it also references the adapter.

So something is stomping on the `local_adapter` variable *and* the `adapters` dictionary, which should never happen after an adapter is bound.  The `adapters` dictionary will be updated when a network number of a directly connected network is "learned" when it receives a `NetworkNumberIs` message, but that doesn't change the `local_adapter` reference, it just replaces the `None` in the dictionary with the network number.  This NPDU is fairly new and relatively unusual and it was added in 0.17.5 so it would be simply dropped in earlier versions and the "to do" at the bottom of the function is deeply suspicious.

The KeyError exception being raised only happens when your application is sending something, so it's probably sending an `IAmRequest` responding to a `WhoIsRequest`, or a ReadPropertyACK.  But the bug could have happened much earlier.

To confirm this suspicion, try turning on debugging for the NSE and send the output to a file giving it lots of space `--debug bacpypes.netservice.NetworkServiceElement:104857600`, that will be just the network layer services but still fairly substantial for large networks.  The adapter has a reference to the NSAP, maybe this can be trapped quickly by putting an `assert adapter.adapterSAP.local_adapter` statement at the bottom of the `NetworkNumberIs()` function.

epaulson, 2019-05-31:
I am emailing you a debug logfile with the error - there is a NetworkNumberIs message coming in, and shortly thereafter the adapters dict has lost its None key and replaced it with a network number. Please let me know if you don't get it. 

I added several asserts into the code; however they were never triggered as far as I can tell:
```
diff --git a/py34/bacpypes/netservice.py b/py34/bacpypes/netservice.py
index bfb44a7..d680703 100755
--- a/py34/bacpypes/netservice.py
+++ b/py34/bacpypes/netservice.py
@@ -1102,16 +1102,19 @@ class NetworkServiceElement(ApplicationServiceElement):

             if _debug: NetworkServiceElement._debug("   - local network learned")
             ###TODO: s/None/net/g in routing tables
+            assert adapter.adapterSAP.local_adapter
             return

         # check if this matches what we have
         if adapter.adapterNet =3D=3D npdu.nniNet:
             if _debug: NetworkServiceElement._debug("   - matches what we have")
+            assert adapter.adapterSAP.local_adapter
             return

         # check it this matches what we know, if we know it
         if adapter.adapterNetConfigured =3D=3D 1:
             if _debug: NetworkServiceElement._debug("   - doesn't match what we know")
+            assert adapter.adapterSAP.local_adapter
             return

         if _debug: NetworkServiceElement._debug("   - learning something new")
@@ -1125,5 +1128,6 @@ class NetworkServiceElement(ApplicationServiceElement):
         # we now know what network this is
         sap.adapters[adapter.adapterNet] =3D adapter

+        assert adapter.adapterSAP.local_adapter
         ###TODO: s/old/new/g in routing tables
```

The error messages that I got at the console were the same as before:
```
(bacpypes176-dbg) pi@raspberrypi:~/bacpypes/samples $ python WhoIsIAm.py --ini ../Base.ini --debug bacpypes.netservice.NetworkServiceElement:debug.txt:104857600
> exception: KeyError(None,)
Traceback (most recent call last):
  File "/home/pi/bacpypes176-dbg/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/app.py", line 372, in indication
    helperFn(apdu)
  File "/home/pi/bacpypes176-dbg/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/service/device.py", line 102, in do_WhoIsRequest
    self.i_am(address=3Dapdu.pduSource)
  File "/home/pi/bacpypes176-dbg/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/service/device.py", line 127, in i_am
    self.request(iAm)
  File "WhoIsIAm.py", line 54, in request
    BIPSimpleApplication.request(self, apdu)
  File "/home/pi/bacpypes176-dbg/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/app.py", line 460, in request
    super(ApplicationIOController, self).request(apdu)
  File "/home/pi/bacpypes176-dbg/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/app.py", line 354, in request
    super(Application, self).request(apdu)
  File "/home/pi/bacpypes176-dbg/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/comm.py", line 574, in request
    self.elementService.sap_indication(*args, **kwargs)
  File "/home/pi/bacpypes176-dbg/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/appservice.py", line 1505, in sap_indication
    self.request(xpdu)
  File "/home/pi/bacpypes176-dbg/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/comm.py", line 574, in request
    self.elementService.sap_indication(*args, **kwargs)
  File "/home/pi/bacpypes176-dbg/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/appservice.py", line 1326, in sap_indication
    self.request(apdu)
  File "/home/pi/bacpypes176-dbg/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/comm.py", line 282, in request
    self.clientPeer.indication(*args, **kwargs)
  File "/home/pi/bacpypes176-dbg/lib/python3.5/site-packages/bacpypes-0.17.6-py3.5.egg/bacpypes/netservice.py", line 292, in indication
    adapter =3D self.local_adapter or self.adapters[None]
```

There are not quite enough timestamps to verify that the WhoIs request happens after the NetworkNumberIs message was processed, unfortunately.

JoelBender, 2019-05-31:
Got the file, thank you!

epaulson, 2019-06-10:
Anything else I can rerun with more debug flags that would be helpful?

JoelBender, 2019-11-17:
A lot of the routing code has been re-written, please upgrade to **0.17.7** and if this is still a problem create a new issue.

closed 2019-11-17 by JoelBender