JoelBender/bacpypes #222: Don't set root logger level to debug

mattdodge, 2018-09-27:
This line is the culprit:
https://github.com/JoelBender/bacpypes/blob/a3c9f1eb4685272dad2ac4d08fedc39e871cac7a/py34/bacpypes/debugging.py#L16

I'm not sure why this needs to happen but it's definitely an anti-pattern when dealing with Python logging. Basically now importing anything `bacpypes` will set any and all Python libraries to log in debug mode. Instead, if you want a "root-level" logger for all bacpypes logging you should rely on the [logging hierarchy](https://github.com/JoelBender/bacpypes/blob/a3c9f1eb4685272dad2ac4d08fedc39e871cac7a/py34/bacpypes/debugging.py#L16) built in to Python's logging.

For example:
```
_bacpypes_root = logging.getLogger('bacpypes')
_bacpypes_root.setLevel(logging.DEBUG)
...
# This logger will inherit the log level from bacpypes logger
_some_logger = logging.getLogger('bacpypes.somelogger')
```

JoelBender, 2018-09-28:
This is a remnant from a _long_ time ago, probably when I started with Python 2.4.  I don't have any record of why I did it that way, but I suspect that unless I added a handler and set the level then I would get `No handlers could be found for logger "__main__"` error output for logging applications.  I wanted to be able to use the same logging feature for attaching handlers to different loggers for both applications and modules.  I think adding a stream handler to the root logger was a way to solve that problem.

For all of the BACpypes modules that create a `ModuleLogger` instance, the module name will include the package name, so for all of the non-BACpypes uses when a `ModuleLogger` is created for a root-like thing then I can add a default handler for warnings and up.

The only thing I use the root logger for BACpypes is debugging which is why it's in the debugging module.  Should it go in the `__init__.py` for the package?

You'll notice right after that is a `if ('--debugDebugging' in sys.argv):` which is another bit of old code that should go away, it is from a time when I looked for options rather than using [optparse](https://docs.python.org/2/library/optparse.html) which has now been deprecated in favor of [argparse](https://docs.python.org/2/library/argparse.html).

Speaking of code that should probably go away, the `commandlogging.py` module never quite worked out the way I expected and should be moved to the sandbox.

mattdodge, 2018-09-28:
Ah ok, that's some useful background, thanks. It's been a while but I do recall some issues with missing logging handlers like that, even with 2.7, so if this dates back to 2.4 I'm not surprised at all.

I'm still not a huge fan of messing with the root logger but doing it only in the debugging module is at least a bit of a tradeoff. The problem is that a lot of the other files `import debugging` so the file basically gets loaded every time.

There's definitely a difficult tradeoff with Python logging between the users that want it to "just work" without any CLI flags or logging JSON files and the users that want to allow full customization of the logging environment. I happen to fall into the latter group, at least in how I'm using `bacpypes` (which is great btw!). 

I haven't confirmed for sure, but I think if you're using `logging.getLogger(__name__)` the loggers will be named something like `bacpypes.object` already. So maybe a quick fix would be to change the `_root` logger reference to this instead?
```python
_root = logging.getLogger('bacpypes')
```

That will still add handlers to the logger, which the user may not want to happen if they have specific logging handler requirements, but at least it's not affecting the global logging namespace.

You also may be able to move that code just in to the `bacpypes_debugging` method in that file, so it's executed only when the decorator is used rather than whenever the file is loaded.

Happy to help with a PR if you have other ideas!

JoelBender, 2018-09-29:
I made a `222-dont-set-root-logger` branch for this and it passes all of my "no barf" tests.  If you are using `logging.config.dictConfig` or `logging.config.fileConfig` I would like to know if you can still take advantage of the `bacpypes.debugging.LoggingFormatter` that has the extra code for "debug contents".  Apparently there's a way to provide the class name of a formatter but I haven't tried.

And I realized that the `commandlogging.py` stuff was the core pieces of being able to create a TCP server that sits quietly in the background but allows you telnet into a running application, turn on debugging and see the output, the disconnect without having to restart the program.

mattdodge, 2018-10-01:
Awesome! We'll give this branch a try with our application to see how it works. And you're right about setting a handler/formatter class path in the dictConfig, we'll make sure the correct formatter works.

I see a couple things in the latest commit on that branch:
1. Can we put the setting of the log level to debug behind the debug flag?
2. Looks like a duplicate/unnecessary assignment of the `logger` object.

I didn't see a PR for it though, if you want to create one I'm happy to move the conversation there to discuss specifics. Thanks again for throwing that together.

JoelBender, 2018-10-02:
(1) I interpret this to mean taking the `root_logger.setLevel(logging.DEBUG)` code out of the debugging module and setting it at the end of the `ConsoleLogHandler()` function in the console logging module?  At that point the function is adding a handler to the logger referenced by `loggerRef` and setting the level, so maybe the level of the root logger should be left unset, subject to the whims of advanced logger types?

(2) Yep, copy/paste error.

I've been operating under the assumption that you would be giving me a PR from your fork, are you expecting a PR from me to go someplace?  I've been creating branches from issues, merging them into the `stage` branch for final testing, then squash merged into `master` for a release.

I remove the set level code, bd6bc3d, try testing again.

JoelBender, 2019-01-22:
This has been merged into `stage` for the next release.

closed 2019-01-22 by JoelBender