JoelBender/bacpypes #224: (Analog,Binay,MultiState)ValueObject Creation / PresentValue “Commandable”

ChristianTremblay, 2018-10-06:
In the actual code, AnalogValueObject and MultiStateValueObject property presentValue are defined as ‘ReadableProperty’ 
For BinaryValueObject property presentValue is defined ‘WritableProperty’
See 
https://github.com/JoelBender/bacpypes/blob/master/py34/bacpypes/object.py#L1016

If we refer to standard (I own 2010), presentValue are ReadableProperty but can be set ‘commandable’ and be modified if so.

I also understand in the code that setting ‘mutable=True’ should allow the property to be written to.

First question would be : why are they not all defined the same in the code (should be ReadableProperty for every one)

Second question : What is the best way to modify the ‘presentValue’ property to mutable as it seem to be the good way to make the property ‘commandable’

In BAC0 I used this : 
```
def _make_mutable(obj, identifier='presentValue', mutable=True):
            for prop in obj.properties:
                if prop.identifier == identifier:
                    prop.mutable = mutable
            return obj
```
I’m wondering if there is a better way to access the property without iterating over all properties.

Tests are passing now
https://github.com/ChristianTremblay/BAC0/tree/develop

JoelBender, 2018-10-06:
The readable/writable difference should match the R/W conformance code column without the special exceptions or combinations which is why the present value of the BinaryOutputObject is writable.  It looks like the BinaryValueObject wasn't mapped correctly back when it was in SourceForge, which was a tedious and error prone process.  There are lots more properties in subsequent additions/addenda and it deserves a comprehensive review.

When a class like BinaryValueObject is "registered" there is a private `_properties` dictionary that maps the property name to the `Property` instance so your for-loop turns into a dictionary lookup.  Note that this is a class attribute, not an instance attribute, so changing the mutable attribute of the `Property` instance is going to change it for all instances of the class which might not be what you want.

In the samples directory is a CommandableMixin.py sample application that is the start of making derived classes like `BinaryValueObjectCmd` which extends a `BinaryValueObject` with a parametric class so instances respond to writing to the priority array correctly. _Eat that code, Java heads!_

If all you want to do is make present value satisfy the "This property is required to be writable
when Out_Of_Service is TRUE." constraint then the class needs to be extended, it's no longer a "schema-like" problem but an "implementation" problem and it belongs in a `bacpypes.local.bv` module, along with implementations of the other variants (and there's 15 footnotes, sigh).  I'm expecting that there will be a similar parametric class definition for the similar object types.  The `CommandableMixin` is going to go in a `bacpypes.local.cmd` module.

ChristianTremblay, 2018-10-06:
As usual, very clear... and I was naive enough to think I had figured it. Hehe! Seems there’s is some work ahead ! I’ll be pleased to help the best I can.

JoelBender, 2018-10-06:
I forgot to point out that you _have_ to use `SomeClassObject._properties` to get to the correct Property instance.  There are a bunch of samples with a `RandomAnalogValueObject` definition which contains a custom property `RandomValueProperty` mapped to the `presentValue` property identifier.  So when the class is registered, `RandomAnalogValueObject._properties['presentValue']` refers to an instance of the `RandomValueProperty` and _not_ to the `ReadableProperty('presentValue', Real)` instance that's in `AnalogValueObject.properties[0]`.

If you go stomping on `AnalogValueObject.properties` you will not change the behavior of `RandomAnalogValueObject` instances.  It's going to get worse with commandable analog value objects that support COV notification and intrinsic reporting with vendor specific properties!

ChristianTremblay, 2018-10-06:
Do you think bacpypes should provide the class already implemented or do you think It is really an implementation issue. If so, we should work out some kind of documentation.
But in any case....
I think I begin to wrap my head around. So everything that would be implemented would fit under `bacnet.local`... because we only care when we generate our own stuff... for the local device to "serve" bacnet points.

There would probably be a mixin for multiple cases :
**bacpypes.local.av.mixins**
 
 * readable_pv (that would be writable only if `out_of_service` is true)
 * commandable_pv
 * intrinsic_reporting
 * support_COV
  
And to create the object we would use
```
from bacpypes.local import av
from bacpypes.local.mixins import commandable_pv, intrinsic_reporting

class SimpleCommandableAV(av,commandable_pv):
    def __init__(args...)
        bla....

class ReportingCommandableAV(av,commandable_pv, intrinsic_reporting):
    def __init__(args...)
        bla....

av0 = av()
av1 = SimpleCommandableAV()
av2 = ReportingCommandableAV()
```

Suis-je dans le champs ?
Am I completely off track ? :-)

JoelBender, 2018-10-07:
No, you're on track.  I was thinking that `bacpypes.local.av` would hold all the various flavors of analog value objects, importing from `bacpypes.local.cmd` for "commandable" and `bacpypes.local.ir` for "intrinsic reporting", `bacpypes.local.cov` for "change of value reporting."  I would probably combine options 3, 6, and 7 together into intrinsic reporting so the Event_Algorithm_Inhibit comes along for the ride.  For the **Value Source Mechanism** (Clause 19.5, Value_Source and Value_Source_Array properties) it's going to be quite a puzzler to mix it in with commandable, you can't have the former without the latter.

My names would be a little different, like `AnalogValueCmd`, `AnalogValueIR`, `AnalogValueCOV`.  Then one massive everything mixed in class, `AnalogValueDogsBreakfast`.  Petit déjeuner de chien? Dîner de chien?  LOL!

[dog's breakfast](https://en.wiktionary.org/wiki/dog%27s_breakfast)

ChristianTremblay, 2018-10-07:
Usually there is a translation for those... but this one has not crossed our frontier ! The closer I get is : “Grand n’importe quoi !” Which is probably as hard to translate back haha!

To keep things clear, I would create folders under ‘bacpypes.local’ 

**bacpypes.local.objects**
 - av.py
     * AnalogValueCmd
     * AnalogValueIR
     * AnalogValueCOV

  - bv.py
     * BinaryValueCmd
     * BinaryValueIR
     * BinaryValueCOV

**bacpypes.local.objects.mixin**
   - cmd.py
   - ir.py
   - cov.py 

I think it would help understand what’s going on... as the mixin will be applied to those objects only.

Application would then use something like : 
```
from bacpypes.local.objects.av import AnalogValueChooseYourFlavor
```

JoelBender, 2019-11-13:
Sitting in `stage` for 0.17.7 release.

closed 2019-11-13 by JoelBender