JoelBender/bacpypes #226: little help with ReadProperty example

brandonfwl, 2018-10-19:
I'm going through the samples and having a little trouble with the ReadProperty.py sample.  I'm not sure what format the console cmd is expecting.  The "help read" provides some guidance but I'm not exactly sure what's expected.  The test device that I have, has the following that I'm trying to read.

addr = 192.168.1.24
instance = 1024
type = multiStateValue
instance = 1 (object instance of the multistatevalue)
property = 'presentValue' (at least that's the property I'd like to read)

Given the information I haven't been able to format the command in a way that works.  Any help would be great.

JoelBender, 2018-10-20:
Part of your confusion might be based on the version of BACpypes you are using.  The older one would use this:
```
> read 192.168.1.24 multiStateValue 1 presentValue
```
however the newer version uses this:
```
> read 192.168.1.24 multiStateValue:1 presentValue
```
Yea, the ':' is pretty subtle.  You might also be confused about the instance number of the device object, a.k.a. the device identifier.  That is used when you want to reference the information about the device itself, like when you want to read the local date or time:
```
> read 192.168.1.24 device:1024 localTime
```
If you think about a BACnet device as a container of objects with an address, you can read (and sometimes write to) properties of objects.  This container always has at least one object, the `DeviceObject` instance, and there is only one of those, all the other objects are something else.

ChristianTremblay, 2020-09-15:
No activity for a long time. Please reopen if you still need help

closed 2020-09-15 by ChristianTremblay