JoelBender/bacpypes #230: WhoIsIAm.py sample not catching all on a large network, Linux

tomschlintz, 2018-11-05:
From a Windows laptop, Bacpypes reports all the Iam responses, but from my Linux box (Advantech 2GHz Celeron, 2G RAM, BIOS V1.10), I only get 114 (sometimes 117) of the 137 device Iam responses.  I'll attach both the WhoIsIAm.py dump and the Wireshark dump:

[whoisiam.txt](https://github.com/JoelBender/bacpypes/files/2549345/whoisiam.txt)
[wireshark.txt](https://github.com/JoelBender/bacpypes/files/2549351/wireshark.txt)

First noticed running from custom code, then decided to run the sample to make sure it's not my mis-implementation.  I'll have more data points once we can run on some other boxes.

JoelBender, 2018-11-06:
My guess is that the application isn't fast enough to read and process each of the incoming packets one at a time, so try [increasing the size of the buffer](https://www.systutorials.com/241303/how-to-enlarge-linux-udp-buffer-size/) and see if that solves it (but 25MB might be overkill).  Adjusting the socket code to keep the buffer empty, putting the packets in a user layer queue and postponing the interpretation, would be more work than I can take on at the present time (turning this into an enhancement request).

tomschlintz, 2018-11-06:
I'll try that.  Thanks for the quick response!

- Tom

JoelBender, 2019-11-17:
Closing this as resolved.

closed 2019-11-17 by JoelBender