JoelBender/bacpypes #232: Unable to perform write request on point object properties of AI/AO/BI/BO

sandhiya88, 2018-11-26:


sandhiya88, 2018-11-26:
Unable to perform write request on point object properties of AI/AO/BI/BO in bacpypes.

I am trying the below command. 
bacnet.write('2002:24 analogInput 10005 presentValue 100') 

gives the error below:
raise NoResponseFromController()
NoResponseFromController

JoelBender, 2018-11-28:
This library doesn't have an API like that, the closest it has would be to pipe that command into one of the sample applications:
```
$ echo "write 2002:24 analogInput:10005 presentValue 100.0" | python samples/ReadWriteProperty.py
```
Note there is a subtle change in the object identifier notation that has a ':' between the object type and instance number.

closed 2018-11-28 by JoelBender

sandhiya88, 2018-11-28:
I tried the command but found that fewproperties are written as only readable in the object.py file and hence, I not able to write to them. Is there any way we can modify them?