JoelBender/bacpypes #234: Reg : BBMD to IP router

preetham94, 2018-12-15:
I am a newbie to BACnet and trying to simulate a BACnet network in a tool and in this process I trying connect two BBMD to a IP router and then boradcast message from one network to another. Is this correct and its enough to use only BBMD to forward the messages from one network to another.

JoelBender, 2018-12-15:
A BBMD is just an IP device that sits on an IP network, there is no "connection" to a router. And it's really only necessary if you want to simulate two IP networks belonging to the same BACnet network, which is probably overkill if you are just starting out.

Let's say you have a test network 192.168.1.0/24 and you want it to look like two BACnet/IP networks with a router between them.  Run the `IP2IPRouter.py` sample application on one PC like this, assuming it has an IP address 192.168.1.99:

    $ python IP2IPRouter.py 192.168.1.99/24:47808 5 192.168.1.99/24:47809 6

_Note that the IP address is the same, but the port number is different._

Now on a different machine, maybe at 192.168.1.55, run a device simulator with the default port 47808, and then even on the same machine you can run other applications like `ReadProperty.py` using the alternate port 47809, or vice versa.  You just can't run two applications on the same machine at the same time if they are using the same port.  I frequently run these as _VirtualBox_ virtual machine instances with a bridged network connection so they are all given separate IP addresses and I can just switch windows to see what's going on.

JoelBender, 2018-12-16:
Yes, as a BACnet router it will listen for both unicast and broadcast messages and forward them between the two networks.  The network layer in BACnet contains Network Protocol Data Units (NPDUs) that are used between devices and routers to discover the path messages should take, and there is a section of application layer messages that help routers forward the message between devices as necessary.

closed 2019-04-11 by JoelBender