JoelBender/bacpypes #235: BACpypes Read Property - YABE

firmwareTeamAmbimat, 2018-12-15:


JoelBender, 2018-12-15:
If the debugging output is from the I-Am that is received from the simulator, it looks like it is coming from 192.168.86.250:65328 which could be a BACnet/IP address on a non-standard port.  If that's true, then your read command would look like this:

    > read 192.168.86.250:65328 analogInput:0 presentValue

The other room controller simulator seems to be at 192.168.86.244:63829, so I'm guessing there is some BBMD-like thing forwarding broadcasts.  The array index is optional of course, unless you are looking for a specific element in a priority array or something like that.

JoelBender, 2018-12-15:
It might be useful for you to run the `Discover.py` sample (which is relatively new) and give it a 'whois' command:

    $ python samples/Discover.py snapshot
    > whois

Give it a chance to discover the stuff it can see, then exit and look at the results:

    $ cat snapshot

If you run the application again with the same snapshot file you'll see the "version number" column update for values that have changed, and if you're running it lots of times and it becomes harder to see the differences, give the application both an input and output file:

    $ python Discover.py snapshot1 snapshot2
    ...
    $ diff snapshot1 snapshot2
    ...

ChristianTremblay, 2020-09-15:
No activity for a long time. If help still required, please reopen

closed 2020-09-15 by ChristianTremblay