JoelBender/bacpypes #237: WhoIsIam.py run function is blocking

AmbimatElectronics, 2019-01-02:
Hello,

We are working on the WhoIsIAm.py and the run function there is blocking the rest of my code and I don't want that to happen. How do I avoid such a situation? How can I selectively turn ON and OFF the execution of the run function in that script.

Thanks

Ambimat

JoelBender, 2019-01-02:
Note: BACnet is a peer-to-peer connection-less protocol, an application can be both a client (initiating requests for data) and a server (responding to requests) at the same time.  Shutting down the `run()` function stops the application from both initiating new requests and responding to the requests from other devices which will eventually cause problems for most applications.

That being said, calling `stop()`, which is also in the `core` module, will set the `running` flag to false and the `run()` function will exit.  You will also need to change `main()` to call `run()` again to resume being able to behave as a BACnet device, and I'm not sure what the impact on your application environment will be when the signal handlers `sigterm` and `sigusr1` are installed multiple times.

JoelBender, 2019-11-17:
Closing this as resolved.

closed 2019-11-17 by JoelBender