JoelBender/bacpypes #239: RandomAnalogValueSleep

bbartling, 2019-01-21:
Hello,

This maybe a Niagara Workbench issue... But would you happen to know why my Betelgeuse controller goes offline right away? It always comes up on a discovery but goes offline with fault code high lighted yellow:

[image]

JoelBender, 2019-01-22:
No, there's a bug someplace.  Run your application with `--debug bacpypes.core` and see if there's an exception it's catching and dropping it.

bbartling, 2019-01-22:
Hi Joel,

So with debug enabled, running the script before a BACnet discovery:
[image]

Then making a discovery:
[image]

And then adding the BACnet device to the network, Betelgeuse is offline instantly.
[image]

[image]

Im running a separate PC on Linux Mint 18.3 and using the Niagara workbench tool on a separate Windows laptop. Please let me know if there is anything I should do differently running the script. Thank you so much for creating such a cool tool!

Ben

ChristianTremblay, 2019-01-22:
Does it react the same if you start a BAC0 instance and discover it with N4
?

If not, something is missing in your code.

Import BAC0
bacnet = BAC0.lite()

Should be enough for the test.

bbartling, 2019-01-22:
BAC0 seems to be working just fine thru N4 discovery... Can BAC0 be used to share a value to a Niagara 4 system? IE, like a calculation in Python share it to the building automation system?

Sorry, Im outwrite confessing I am not a software engineer testing my BACnet creations ;) but a BAS technician experimenting with Python DIY computer vision project

[image]
[image]

bbartling, 2019-01-22:
Would you have any tips for trouble shooting BAC0 going offline to Niagara? Are there any tuning parameters or firewall issues on a PC that would make BAC0 go offline?

Its offline in Niagara but thru the console it looks fine:

[image]

jscott3201, 2019-01-22:
What’s the fault status in Niagara reading on the offline. Console logs
from the platform? Niagara by default uses polling and if it can respond
fast enough it may fault out. Using actions > Ping on the device might
bring it back up.

bbartling, 2019-01-22:
Here's a platform screenshot. It looks like BAC0 is just a transaction timeout.. Can polling adjustment fix that? Pinging didn't bring anything back to life either...

Not sure about the Betelgeuse error shown before BAC0
[image]

jscott3201, 2019-01-22:
Something wrong in your device setup code. If I recall correctly pings
check the Device Type (8?). Top error shows it can’t decide the property
correctly. Last Ok time being bull shows it never really pulled the device
in properly. Device pulls in successfully to Niagara the Last Ok Timestamp
should be filled.

bbartling, 2019-01-22:
@jscott3201, are you referring to BAC0 or BACpypes where I may have something setup/configured wrong? Thanks

jscott3201, 2019-01-22:
BACpypes side likely. Repo or snippet of your code and .ini file? Most
device props get set off ini file I believe.

bbartling, 2019-01-22:
In the bacpypes/samples directory I am working with running the RandomAnalogValueSleep.py

[image]

There is also another INI with a ~ 
-sorry I don't know the difference but they look almost exactly the same
[image]

jscott3201, 2019-01-22:
I’m on the Subway but the sound of the sample name may be the giveaway
here. If that Value Sleep code is calling core.sleep(), it’s putting the
python thread to sleep which means the server is technically offline and
can’t respond to the messages. I’ll peek at the sample and post if I see
that line. core.run() starts the BACnet python thread which is your device
BAC0 in this case.

jscott3201, 2019-01-22:
This line: # sleep a little
        time.sleep(args.sleep)

Not sure off quick glance but that might be putting the thread to sleep. Pass in 60s for that CLI argument. Might keep it awake enough to verify that is the cause on the N4 side.

JoelBender, 2019-01-22:
The sleep time defaults to 100ms to slow things down _a little_, like it's a low powered device on a small network with hardly enough spare CPU cycles to maintain the correct time.  You can gradually increase that time, we have some virtual devices in a gateway that then communicates over 900MHz radio links where the response time can be more than that, but it's pretty rare.

Passing 60 _seconds_ means that whenever the present value of one of its Analog Value Object's the application will be locked up doing nothing.  Requests will get backlogged, multiple attempts will be made, and as you've seen, most clients will give up.  Some will give up a little and try again by themselves, some will give up until you tell them to try again with a supervisory command/interface.

If there was a separation between the application layer processing requests and the communications layer for sending/receiving traffic, then in theory the communications could abort with a **DEVICE_BUSY** error code (Clause 18.1 Error Class - DEVICE).

JoelBender, 2019-01-22:
@jscott3201 That sample application doesn't use any threads, the `sleep()` call stops everything.  On a busy network the OS layer UDP socket buffer will fill up an it will start dropping packets.  Thank you and @ChristianTremblay for picking up the conversation.

ChristianTremblay, 2019-01-23:
Running a local N4 station (4.6), I've imported 2 different BAC0 devices, one running on ubuntu, another one on RPi... 
Online, no problem.
Wireshark is clean... 

Bacpypes 0.17.6 and 0.17.5

[image]

I've simply imported some string points...
[image]

Niagara4 BACnet IP configuration
[image]

In 2 different subnets as you can see, no BBMD as the Jace will unicast the manually added device in the other subnet.

So if BAC0 is failing on your side, it's probably a network OR a Niagara issue.
If using a custom bacpypes, there is a parameter somewhere.

Disclaimer : Using BAC0 here only as it is a lego app built with a bunch of examples of bacpypes. Any bacpypes should behave the same if configured in a similar way.

bbartling, 2019-01-23:
Thanks for the help all, I think I am doing something wrong in Bacpypes. I installed BAC0 on raspberry pi and its working just fine how @ChristianTremblay is demonstrating above with the N4 integration..

closed 2019-01-23 by bbartling