JoelBender/bacpypes #240: TestTimeMachine.test_recurring_task_5 fails

icrde, 2019-01-24:
self = <tests.test_utilities.test_time_machine.TestTimeMachine testMethod=test_recurring_task_5>
    
        def test_recurring_task_5(self):
            if _debug: TestTimeMachine._debug("test_recurring_task_5")
    
            # create a function task
            ft = SampleRecurringTask()
    
            # reset the time machine, install the task, let it run
            reset_time_machine(start_time="2000-01-01")
            ft.install_task(86400.0 * 1000.0)
            run_time_machine(stop_time="2000-02-01")
    
            # function called every day
    >       assert len(ft.process_task_called) == 31
    E       AssertionError: assert 30 == 31
    E        +  where 30 = len([946771200.0, 946857600.0, 946944000.0, 947030400.0, 947116800.0, 947203200.0, ...])
    E        +    where [946771200.0, 946857600.0, 946944000.0, 947030400.0, 947116800.0, 947203200.0, ...] = <tests.test_utilities.test_time_machine.SampleRecurringTask object at 0xf3614050>.process_task_called

    tests/test_utilities/test_time_machine.py:278: AssertionError

Looking into it, it seems that the first event is scheduled for 0.000001 seconds into 2000-01-02, and the 31st event is scheduled for 0.000001 seconds into 2000-02-01, which is after the time machine shuts down.

JoelBender, 2019-01-25:
Please make a copy of `test_script.sh`, call it `test_time_machine.sh`, and stuff this in the middle where the BACPYPES_DEBUG environment file is exported:
```
# add the modules or classes that need debugging and redirect the output to the file
export BACPYPES_DEBUG=" \
    bacpypes.task:$bugfile \
    tests.time_machine:$bugfile \
    tests.test_utilities.test_time_machine:$bugfile \
    "
```
You are correct, the first time the task runs it generates debugging output like this:
```
DEBUG:tests.test_utilities.test_time_machine.SampleRecurringTask:process_task @ 946771200.0 01/02/00 00:00:00
```
The last one should be `process_task @ 949363200.0 02/01/00 00:00:00` because the time limit is at 949381200.0.  I suspect that the microsecond floating point "wiggle" that keeps a lot of these floating point comparisons off isn't working correctly, but so far I can't reproduce it.

JoelBender, 2019-11-17:
Closing this as resolved, please upgrade to **0.17.7** and if there's a problem create a new issue.

closed 2019-11-17 by JoelBender