JoelBender/bacpypes #243: Syntax Error when running bacpypes on Python 2.6.6

quest-gmulcahy, 2019-02-15:
When trying to run bacpypes via Python version 2.6.6 (the default repo version of CentOS6) the following error resulted:
```
python ./samples/Tutorial/WhoIsIAm.py Traceback (most recent call last): File "./samples/Tutorial/WhoIsIAm.py", line 11, in <module>
from bacpypes.debugging import bacpypes_debugging, ModuleLogger
File "/usr/lib/python2.6/site-packages/bacpypes/init.py", line 29, in <module>
from . import comm
File "/usr/lib/python2.6/site-packages/bacpypes/comm.py", line 416
self.terminals = {k:Switch.TerminalWrapper(self, v) for k, v in terminals.items()}
^
SyntaxError: invalid syntax"
```
I'm going to try to update to a newer version of Python, but adding this item as a reference if anyone else runs into the problem.

quest-gmulcahy, 2019-02-15:
Here was an initial response from Joel, just to provide additional context:

Joel Bender
@JoelBender
11:52
That syntax is called “dictionary comprehension” and they were introduced in Python 2.7 (PEP 274). I lost my machine that could run Python 2.5, so this is a little like programming blind. That code is a copy/paste from the 2.7 version, so it needs to be updated.
Please create an issue for this and I’ll clean it up. In the mean time, please see what you can do on your platform to support Python2.7.

JoelBender, 2019-02-15:
Check out the `243-syntax-error` branch and see if it works, I'll merge it into the next release.

quest-gmulcahy, 2019-02-19:
Unfortunately, I'm not that familiar with git yet, but I manually edited the files to mimic the changes represented in the 243-syntax-error branch (appservice.py & comm.py). 

This seems to have resolved the errors after also installing argparse
`pip install argparse`

I'm seeing a deprecation warning, but it isn't blocking. 
```
 python samples/WhoIsIAm.py
/usr/lib/python2.6/site-packages/bacpypes/app.py:235: DeprecationWarning: local address at the application layer deprecated
  DeprecationWarning,
```
As noted above this utility is working without any issues with Python3.4 installed.
Thanks for your time.

closed 2019-04-11 by JoelBender