JoelBender/bacpypes #244: Raspberry PI as BACnet Controller

kazemiprince, 2019-02-20:
Hello all.
I'm trying to control an AHU (Air  Handling Unit) with Raspberry Pi (GPIOs) and communicate with Raspbbery via BACnet for monitoring and changing set-point and ...
after lots of search on internet, I found that the BACnet stack that I have to use is BACpypes.
but I'm totally confused and I couldn't find any sample app to work as server.
I found one server sample with C and it's work with Raspberry GPIO too, but I'm looking for Python BACnet stack.
it's highly appreciated if any one send me simple sample app (not console) to see how to use BACpypes as server.

JoelBender, 2019-02-21:
Are you going to be using the [gpiozero](https://gpiozero.readthedocs.io/en/stable/) library?  I would expect a "minimum viable sample" application to have buttons related to binary input and output objects.  I'll take one of the sample applications and put in some markers where you can add some code.

JoelBender, 2019-02-21:
The sample server application in the `stage` branch does not import the gpiozero library, so it can't do anything "real" until you can test it on the real thing.  I also do not understand the relationship between the `pause()` function from the signal library and how that impacts the fact that BACpypes applications run their own `select()` loop in the core.

tjohnsonhvac, 2019-02-21:
I have a rpi3 I can look at testing. I haven’t messed with the gpio stuff for the rpi3 in python. I’ll do some reading tonight.

tjohnsonhvac, 2019-02-21:
So the binaryio.py needs to be tested on an rpi?

JoelBender, 2019-02-21:
(1) add a `from gpiozero import Button, LED` to the application, (2) remove the mock classes, (3) make a call to `obj._button.is_pressed` and map that to **active** or **inactive**, (4) update the BO code to check the value and call `obj._led.on()`

ChristianTremblay, 2019-02-21:
Dirty hack using BAC0... started from this https://github.com/ChristianTremblay/BAC0/issues/126
```
import BAC0
import time
import RPi.GPIO as GPIO

# GPIO Used
LED = 26

# Pin Setup:
GPIO.setmode(GPIO.BCM)   # Broadcom pin-numbering scheme.
GPIO.setwarnings(False)
GPIO.setup(LED, GPIO.OUT)

from BAC0.tasks.RecurringTask import RecurringTask 
from bacpypes.primitivedata import Boolean
from BAC0.core.devices.create_objects import create_BO

def makeLEDasBO():
    obj = bacnet.this_application.get_object_name('MyLED')
    value = obj.ReadProperty('presentValue')
    if value == 'active':
        GPIO.output(LED,True)
    else:
        GPIO.output(LED,False)
     
def _make_mutable(obj, identifier='presentValue', mutable=True):
    """ 
    This function is not the way to go as it changes the class
    property...As bacpypes issue #224, it will need a lot of work
    """
    for prop in obj.properties:
        if prop.identifier == identifier:
            prop.mutable = mutable
    return obj

bacnet = BAC0.lite()
bacnet.this_application.add_object(_make_mutable(create_BO(oid=0, name='MyLED', pv=0)))
print(bacnet)
task = RecurringTask(makeLEDasBO,delay=1)
task.start()

while True:
    obj = bacnet.this_application.get_object_name('MyLED')
    print('Running and value is {}'.format(obj.presentValue))
    time.sleep(10)
```

A Red Led is connected to GPIO 26
Another instance connects to this device using : 
```
import BAC0
bacnet = BAC0.lite(port=47809) # using another port to make this work on the same RPi
dev = BAC0.device('192.168.210.21',3056177,bacnet)
dev['MyLED'] = True
# Led Is now ON
dev['MyLED'] = False
# Led is now OFF
```

This is more a "Remote IO" type of example... but it is a funny 30 minutes hack

ChristianTremblay, 2019-02-21:
In BACnet, you must change your mindset about client and server. A bacpypes application act as both. It’s a device. 

BAC0 is just a wrapper around bacpypes. It’s built from many Joel’s example. Nothing more than a Lego game... It shows here the concept that your bacpypes app will need to be given objects and it will be configured to react to BACnet requests. 

As you can see, the first part of the example shows an instance to which I add a binary output : « MyLed » 
This output drives a led on GPIO 26 

The second example connects to the first one and controls the led giving orders to it. 

As I used two different port and i’m Connecting directly to the first device using another instance, you will not see Whois iam. Those are broadcast messages not seen if using different ports. This is another topic. The example here is only about controlling RPi GPIO using BACnet.

kazemiprince, 2019-02-21:
But in my case, RPI should be reachable by another machine such as my SCADA software on my laptop, therefore RPI should reply to "who-is" message from my laptop (BACnet Discovery tool or any SCADA software) with "iam" message including device description and point list as well.
would you please let me know how to upgrade your code to play in this role?

JoelBender, 2019-02-21:
This problem almost always is because the IP networking configuration that you provided the application doesn't have the CIDR network size set.  Please check your subnet mask for the network size (for example, if your mask is **255.255.255.0** then the network size is 24 bits) and use that in the address setting, i.e., **192.168.0.12/24**.  Your BACnet discovery tool will be sending out Who-Is broadcast requests to **192.168.0.255** and BACpypes has to know to listen for those PDUs.  If you don't have any subnet specified, it will listen for "old school" broadcasts going to **255.255.255.255** and there aren't any.

tjohnsonhvac, 2019-02-22:
^^ I had many issues with the same thing with my RPI3 router not responding to broadcasts... subnets are everything..
I finally got my other RPI up and running, I'm testing the program now.

JoelBender, 2019-02-22:
Pull request #245 accepted, thank you.

kazemiprince, 2019-02-22:
Thanks for your reply but subnet mask is right and still I can not see reply in who-is request.
here is BACpypes.ini : 
`[BACpypes]
objectName: Betelgeuse
address: 192.168.137.189/24
objectIdentifier: 599
maxApduLengthAccepted: 1024
segmentationSupported: segmentedBoth
vendorIdentifier: 15
foreignBBMD: 128.253.109.254
foreignTTL: 30
`
RPI is connected thru a lan cable directly to my laptop, my laptop IP is 192.168.137.1, and RPI IP is 192.168.137.189 with subnet mask 255.255.255.0 on both side.
and here is report from WireShark after seraching for device via BACnet Discovery tool:
[image]

as you can see in image, RPI is replying to who-is, but not with iam.
sorry to disturb you, but I'm confused why?!

JoelBender, 2019-02-23:
There is something strange going on, please provide the contents of the Who-Is request in [3], including the BVLL header.  If the response from the RPi is the same in [4] as it is in [22], then the BVLL header in [3] is weird.  You should be able to run the `WhoIsIAm.py` sample application on the laptop, put in the simplest `whois` command, see the result broadcast on the LAN (sent to 192.168.137.255) with an Original Unicast BVLL message on the wire.  And the response as well, of course, but first make sure the request is well formed.

kazemiprince, 2019-02-23:
Here is content of Who-IS : 
[image]
and BVLC from RPI to Laptop:
[image]
BVLC from Laptop to RPI
[image]
my laptop is running on windows 7, and bacpypes is not installed on it, but I will try with another VM with Ubantu and I will update you, but just for your information if I search with BACnet Discovery tool from my laptop, and send "iam" with WhoIsIAm.py from RPI frequently during this search, I will see the RPI in BACnet discovery tool, and it means "who-is" message from BACnet Discovery tool (Laptop) is well fragmented. Also if I run Bacnet-Stack in RPI, again I can found RPI in Bacnet Discovery Tool search result that it means again "Who-Is" from Laptop is in correct way.
note: I copy and paste your sample code without any change and just run it with python3 and python with same result.

kazemiprince, 2019-02-24:
so, I just try some another test and found some difference when I use Bacnet-Stack, and sample app with Python and even WhoIsIAm.py
In Bacnet-Stack and WhoIsIAm.py, the BVLC length is 48, for both side (Laptop to RPI and reverse)
In the sample app that you prepared, from Laptop to RPI is 48 but from RPI to Laptop is 60.
how? why? should be same or it's not the problem!

JoelBender, 2019-02-24:
The problem is in the configuration of the client, the device making the request.  Look in the BVLL header and you'll see `Distribute-Broadcast-To-Network` where it should be `Original-Broadcast`.  The client is treating your server as a BACnet Broadcast Management Device (BBMD) and behaving as a registered foreign device.  When it receives the negative acknowledgement (NACK) it thinks "oh, must be my registration has exprired, let me try registering again, which also fails.

Your device is the simplest BACnet/IP device, it should only be sending **Orginal-Broadcast** or **Original-Unicast** messages.  You have a really simple network, you don't have two IP networks that you are attempting to make one BACnet network, so a vast majority of the rest of the BVLL messages aren't necessary.

So to answer your other question; yes, the size being different is an indication that there is a difference in behavior, but it's "it's not the size that matters" (LOL) what's _in_ the packet that says what's really going on.  Check for a setting that has something like "foreign device" with an IP address, socket, and time-to-live and make sure it is unchecked/disabled.

kazemiprince, 2019-02-25:
You're the best!
My mistake was that I set RPI IP address as BBMD address in Bacnet Discovery tool:
[image]

Just after I removed the BBMD, I could find it from Laptop.
Thanks a lot.
[image]

JoelBender, 2019-02-26:
Happy hacking!

closed 2019-02-26 by JoelBender