JoelBender/bacpypes #247: Read object list

firmwareTeamAmbimat, 2019-02-28:
I run whois than i get the "remotestation " instead of  "Address".
I try to readobject of remotestation device but i can not read it. please give me solution how can i read the object of remotestation device.

Thank you.
[image]

JoelBender, 2019-03-01:
A "remote station" is a device that is on a BACnet network different than the one you are directly connected to and therefore you need a router to help forward the message along.  The `RemoteStation` object contains both the network number and the MAC address of the BACnet device on that other network, and you can use the same syntax `20021:22` as the destination address of your unicast traffic like read requests.  So, for example:

     > read 20021:22 device:2200 objectName

Will return the object name.

AmbimatElectronics, 2019-03-05:
Hello Joel,

We are using a Linux terminal to run the script as suggested. 
[image]

After running the script there is no response back.

Are we missing something?

Neel.

firmwareTeamAmbimat, 2019-03-05:
Hello,

We tried once again and looks like we got a response but just once and then the terminal is stuck waiting for a response.

We tried couple of times after that and we are still not getting a response back.

Let us know what we are missing.

Firmware Team
firmwareteam@ambimat.com

JoelBender, 2019-03-06:
There is something amiss with your configuration.  If you can run Wireshark (with a capture filter on just BACnet traffic `udp port 47808`) and send it to me, along with your INI configuration file and output from `$ ifconfig` or `$ ip addr show` then I can probably figure out what's going on.

AmbimatElectronics, 2019-03-06:
Hi Joel,

Here are some conclusions that we have drawn. We tried running the ObjectName request for 4 different devices.

[image]

At the fourth device call the code gets stuck waiting for response. The line number is #74 in the file 

https://github.com/JoelBender/bacpypes/blob/master/samples/ReadProperty.py

The same devices when called in a different order will respond for the first few times but not after that.

[image]

Here is the requested BACpypes.INI file

[image]

and the output of the ifconfig file

[image]

Please suggest your thoughts.

Thanks

JoelBender, 2019-03-09:
I suspect this is because there is a mismatch between the content in the request and the response because they are running in separate threads.  The console is calling the application `request_io()` in its thread but it should actually be called in the main thread (which is the primary thread in the sample applications). Please pull the `stage` branch and run the application again.  Thank you for your patience and persistence.

AmbimatElectronics, 2019-03-13:
Hi Joel,

Thank you for your suggestion on the threading. We got that working just fine now.

Such said, we are now faced with another challenge. Reading objects of the devices.

[image]

[image]

For the MSTP/IP router we are NOT able to get the object list.

Same is the case with Device ID 2000. See screenshot below:

[image]

Please advice.

Thanks

ChristianTremblay, 2020-09-15:
No activity for a long time. If help still required, please reopen

closed 2020-09-15 by ChristianTremblay