JoelBender/bacpypes #251: BASRouter from Contemporary Controls

AmbimatElectronics, 2019-03-05:
Hello,

Here is a device which we are looking to purchase and we are going to be using your library to communicate with the device. 

Can you confirm if you or someone from the community has been able to connect and retrieve data from the devices via this BASRouter?

Thanks

Neel.
[image]
[image]

ChristianTremblay, 2019-03-05:
Works well

AmbimatElectronics, 2019-03-05:
Thank you.

Can you please suggest how do we access properties like the Object name on a network like this?

The Router is on 2300 and the other devices are connected subsequently on 30100, 30103, 30115 and so on.

Firmware Team
firmwareteam@ambimat.com

ChristianTremblay, 2019-03-05:
You simply point the controller you need using network:mac syntax.

Ex for MSTP network 2300 and controller at MAC address 4 giving device ID
4000

2300:4 device 4000 objectName

JoelBender, 2019-03-06:
The picture on the front of the router doesn't say which network is the IPv4 network and which one(s) are MS/TP.  When you have everything connected, run the `WhoIsRouter.py` sample application and give it the `wirtn *` command, you will see a response from the other devices on your local network that are routers and the reachable networks.  You can also run the `WhoIsIAm.py` sample, send out a global broadcast request for all of the devices on your network by giving it `whois`, and look carefully at the responses from the devices for their BACnet address.  For ones coming through the router, they will look like what Christian said; a network number, a colon, and a MAC address.

AmbimatElectronics, 2019-03-06:
Dear Joel,

Please see the images attached with this email.

The one with the Device ID#2 is an ‘ONLY’ BACnet IP device.
The one with the Device ID#2300 is a BACnet IP Router.
The rest of the devices shown on the image are the BACnet MS/TP devices. We cross verified the same with the client as well.

[image]

[image]

[image]

Based on the three images (Yabe, whois and wirtn *) we can ascertain that there are 7 devices connected to the router #2300. However the devices addresses are different between the wirtn * command and the whois / Yabe. For example: We have a 2000 device address above but we don’t see a 10020 any where. Also there is no 50023 in the whois response but it is present in the wirtn * command.

Thoughts?

JoelBender, 2019-03-07:
The `WhoIsRouterToNetwork` message is a **network layer message** used to find the topology of the network.  The response contains network numbers, not device identifiers or device addresses.  You can directly determine the BACnet network topology using these messages, or infer the topology as a result of other **application layer messages** like I-Am.

You are correct, the I-Am messages in response to you Who-Is show that there are 7 devices connected on the "other side" of the router, and they are on three of the seven networks the router knows about;  network 10023 has one device with something that looks like an Ethernet MAC address, network 20021 has four devices, and network 50001 has two devices.  The other networks like 50023 seem to have no devices on them.

In BACnet the device identity (such as 2000, 2100, 2200, in your example) is designed to be persistent and unique in the entire BACnet network.  The corresponding network address (which consists of two pieces, a network number and a MAC address) is the "path" to send messages to the device.

Your read request for the object name needs these identifiers in different parts of the command:
```
    > read 20021:5 device:2500 objectName
```
The first parameter is the BACnet network address, the second is the device object identifier, and the third is the property.  Try this one too:
```
    > read 50001:3 device:30103 objectName
```
Both of these requests will go to the router which will forward them on to different networks, and route the response back to you.

AmbimatElectronics, 2019-03-07:
Dear Joel,

Thank you for the detailed understanding here.

We get where you are coming from.

Ambimat Electronics
support@ambimat.com

closed 2019-04-11 by JoelBender