JoelBender/bacpypes #252: Reg: Add/delete objects

preetham94, 2019-03-05:
I am thinking of a client application which could add or delete objects to a server. Could you please guide me on how to do that or is there an example for this task ?

JoelBender, 2019-03-06:
There is currently no sample application for this problem.  The `DeleteObjectRequest` confirmed service is about as simple as it can get, you simply give it the object identifier.  The `CreateObjectRequest` is more complicated because it involves building a sequence of property/value pairs.  The `value` element in the `PropertyValue` sequence is an `Any` which could mean you need to build somewhat complicated nested pieces of stuff.  You will also need to know exactly what the server requires and will support for the object type, and maybe the server is at a particular BACnet revision level that doesn't support some "required" properties, recognize some "optional" ones, etc.

If you have a specific type of object in mind (and maybe not a schedule object to start!) then I can show you how to build a request as a "template" and you can tweak it.  Note that BACpypes applications as a server do not support these services.

ChristianTremblay, 2020-09-15:
Not planned yet, we'll close for now.

closed 2020-09-15 by ChristianTremblay