JoelBender/bacpypes #253: Discover.py OrderedDict not defined

quest-gmulcahy, 2019-03-06:
rpm 5531 device:5531  lastRestoreTime
ERROR:__main__.DiscoverConsoleCmd:exception: NameError("name 'OrderedDict' is not defined",)
Traceback (most recent call last):
  File "samples/Discover.py", line 755, in do_rpm
    dict_contents = value.dict_contents(as_class=OrderedDict)
NameError: name 'OrderedDict' is not defined

when doing a read for 'all' or 'lastRestoreTime' there seems to be an issue handling an OrderedDict 

The property seemed to be handled correctly in the 'samples/ReadPropertyMultiple.py' script.

JoelBender, 2019-03-07:
Thank you, check this again and close the issue if it now works for you.

quest-gmulcahy, 2019-03-07:
Thanks, this seems to work. Sorry about the inconvenience, not quite familiar enough with the library to fix that type of issue myself yet.

closed 2019-03-07 by quest-gmulcahy