JoelBender/bacpypes #254: Handling 'StructuredObjectList' in Discover.py and ReadPropertyMultiple.py

quest-gmulcahy, 2019-03-07:
'''
rpm 3000 device:3000 structuredObjectList
ERROR:__main__.DiscoverConsoleCmd:exception: InvalidTag('object identifier application tag required',)
Traceback (most recent call last):
  File "samples/Discover.py", line 752, in do_rpm
    value = propertyValue.cast_out(datatype)
  File "/root/bacnet/tutorial_env/lib64/python3.4/site-packages/bacpypes/constructeddata.py", line 1327, in cast_out
    helper.decode(t)
  File "/root/bacnet/tutorial_env/lib64/python3.4/site-packages/bacpypes/constructeddata.py", line 886, in decode
    helper = self.subtype(tag)
  File "/root/bacnet/tutorial_env/lib64/python3.4/site-packages/bacpypes/primitivedata.py", line 1652, in __init__
    self.decode(arg)
  File "/root/bacnet/tutorial_env/lib64/python3.4/site-packages/bacpypes/primitivedata.py", line 1736, in decode
    raise InvalidTag("object identifier application tag required")
bacpypes.errors.InvalidTag: object identifier application tag required
```
This error also comes up when performing a read on 'all' properties of a device.

JoelBender, 2019-03-08:
Please do a pull, I added some debugging, and run it again with debugging turned on for the command, `--debug __main__.DiscoverConsoleCmd`.  That property is just an array of object identifiers according to 2016, but maybe it has changed.

quest-gmulcahy, 2019-03-08:
Please see below:
```
 rpm 3000 device:3000 structuredObjectList
DEBUG:__main__.DiscoverConsoleCmd:do_rpm ['3000', 'device:3000', 'structuredObjectList']
DEBUG:__main__.DiscoverConsoleCmd:    - devid: 3000
DEBUG:__main__.Snapshot:get_value 3000 '-' 'address'
DEBUG:__main__.DiscoverConsoleCmd:    - addr: '40001:0'
DEBUG:__main__.DiscoverConsoleCmd:    - obj_id: ('device', 3000)
DEBUG:__main__.DiscoverConsoleCmd:    - prop_id: 'structuredObjectList'
DEBUG:__main__.DiscoverConsoleCmd:    - request: <bacpypes.apdu.ReadPropertyMultipleRequest(14) instance at 0x7f46a1f57908>
    <bacpypes.apdu.ReadPropertyMultipleRequest(14) instance at 0x7f46a1f57908>
        pduDestination = <Address 40001:0>
        pduExpectingReply = 1
        pduNetworkPriority = 0
        apduType = 0
        apduService = 14
        listOfReadAccessSpecs
            [0]                objectIdentifier = ('device', 3000)
                listOfPropertyReferences
                    [0]                        propertyIdentifier = 'structuredObjectList'
        pduData = x''
DEBUG:__main__.DiscoverConsoleCmd:    - iocb: <bacpypes.iocb.IOCB(3) instance at 0x7f46a1f6a2e8>
    <bacpypes.iocb.IOCB(3) instance at 0x7f46a1f6a2e8>
        args = (<bacpypes.apdu.ReadPropertyMultipleRequest(14) instance at 0x7f46a1f57908>,)
        kwargs = {}
        ioState = 0
        ioComplete = <threading.Event object at 0x7f46a1f57080>
        ioCallback = []
        ioPriority = 0
DEBUG:__main__.DiscoverConsoleCmd:    - apdu: <bacpypes.apdu.ReadPropertyMultipleACK(14,3) instance at 0x7f469f9fd358>
    <bacpypes.apdu.ReadPropertyMultipleACK(14,3) instance at 0x7f469f9fd358>
        pduSource = <Address 40001:0>
        pduExpectingReply = False
        pduNetworkPriority = 0
        apduType = 3
        apduSeg = False
        apduMor = False
        apduService = 14
        apduInvokeID = 3
        listOfReadAccessResults
            [0]                objectIdentifier = ('device', 3000)
                listOfResults
                    [0]                        propertyIdentifier = 'structuredObjectList'
                        readResult
                            propertyValue
                                    <bacpypes.primitivedata.Tag(null) instance at 0x7f469f9fd588>
                                        tagClass = 0 application
                                        tagNumber = 0 null
                                        tagLVT = 0
                                        tagData = ''
        pduData = x''
DEBUG:__main__.DiscoverConsoleCmd:    - objectIdentifier: ('device', 3000)
DEBUG:__main__.DiscoverConsoleCmd:    - propertyIdentifier: 'structuredObjectList'
DEBUG:__main__.DiscoverConsoleCmd:    - propertyArrayIndex: None
DEBUG:__main__.DiscoverConsoleCmd:    - datatype: <class 'bacpypes.constructeddata.ArrayOf.<locals>.ArrayOf'>
ERROR:__main__.DiscoverConsoleCmd:exception: InvalidTag('object identifier application tag required',)
Traceback (most recent call last):
  File "samples/Discover.py", line 754, in do_rpm
    value = propertyValue.cast_out(datatype)
  File "/root/bacnet/tutorial_env/lib64/python3.4/site-packages/bacpypes/constructeddata.py", line 1327, in cast_out
    helper.decode(t)
  File "/root/bacnet/tutorial_env/lib64/python3.4/site-packages/bacpypes/constructeddata.py", line 886, in decode
    helper = self.subtype(tag)
  File "/root/bacnet/tutorial_env/lib64/python3.4/site-packages/bacpypes/primitivedata.py", line 1652, in __init__
    self.decode(arg)
  File "/root/bacnet/tutorial_env/lib64/python3.4/site-packages/bacpypes/primitivedata.py", line 1736, in decode
    raise InvalidTag("object identifier application tag required")
bacpypes.errors.InvalidTag: object identifier application tag required
ERROR:__main__.DiscoverConsoleCmd:exception: InvalidTag('object identifier application tag required',)
Traceback (most recent call last):
  File "samples/Discover.py", line 754, in do_rpm
    value = propertyValue.cast_out(datatype)
  File "/root/bacnet/tutorial_env/lib64/python3.4/site-packages/bacpypes/constructeddata.py", line 1327, in cast_out
    helper.decode(t)
  File "/root/bacnet/tutorial_env/lib64/python3.4/site-packages/bacpypes/constructeddata.py", line 886, in decode
    helper = self.subtype(tag)
  File "/root/bacnet/tutorial_env/lib64/python3.4/site-packages/bacpypes/primitivedata.py", line 1652, in __init__
    self.decode(arg)
  File "/root/bacnet/tutorial_env/lib64/python3.4/site-packages/bacpypes/primitivedata.py", line 1736, in decode
    raise InvalidTag("object identifier application tag required")
bacpypes.errors.InvalidTag: object identifier application tag required
```

JoelBender, 2019-03-09:
There are some properties that have a NULL value, like _Relinquish_Default_ and _Current_Command_Priority_, and while I don't see it as a valid option for the _Structured_Object_List_ property of the Device Object, the application should be able to handle it gracefully.  Making a special case in this application will help with the immediate problem, but this will probably be an issue with other applications.