JoelBender/bacpypes #255: Discover.py error when sorting data for outfile

quest-gmulcahy, 2019-03-07:
```
Traceback (most recent call last):
  File "samples/Discover.py", line 840, in <module>
    main()
  File "samples/Discover.py", line 837, in main
    snapshot.write_file(args.outfile)
  File "samples/Discover.py", line 158, in write_file
    data.sort()
TypeError: unorderable types: str() < int()
```

JoelBender, 2019-03-08:
The "key" for the data is the combination of the device identifier, object identifier, and property identifier, and at least one of those is in the wrong format.  Please turn on debugging with `--debug __main__.Snapshot` and look for the `upsert()` calls.  All of the parameter combinations should be the same datatype and depending on which one is wrong will tell me what needs help.  I'm guessing that a property identifier that is a part of some object is proprietary or my application doesn't recognize it yet.

quest-gmulcahy, 2019-03-08:
My intention was to read all properties from a device and get the data via outfile to review all the device properties, then eventually object properties, below is a small portion of the debug data. 
```
python3 samples/Discover.py infile outfile --debug __main__.Snapshot
DEBUG:__main__.Snapshot:__init__
DEBUG:__main__.Snapshot:read_file 'infile'
> rpm 6400 analogInput:5 all
DEBUG:__main__.Snapshot:get_value 6400 '-' 'address'
objectIdentifier: ('analogInput', 5)
DEBUG:__main__.Snapshot:upsert 6400 'analogInput:5' 'objectIdentifier' "('analogInput', 5)"
DEBUG:__main__.Snapshot:    - new key
objectType: analogInput
DEBUG:__main__.Snapshot:upsert 6400 'analogInput:5' 'objectType' 'analogInput'
DEBUG:__main__.Snapshot:    - new key
objectName: CT
DEBUG:__main__.Snapshot:upsert 6400 'analogInput:5' 'objectName' 'CT'
DEBUG:__main__.Snapshot:    - new key
description: CT
DEBUG:__main__.Snapshot:upsert 6400 'analogInput:5' 'description' 'CT'
DEBUG:__main__.Snapshot:    - new key
1033: ?
DEBUG:__main__.Snapshot:upsert 6400 'analogInput:5' 1033 '?'
DEBUG:__main__.Snapshot:    - new key
presentValue: 60.0
DEBUG:__main__.Snapshot:upsert 6400 'analogInput:5' 'presentValue' '60.0'
DEBUG:__main__.Snapshot:    - new key
1035: ?
DEBUG:__main__.Snapshot:upsert 6400 'analogInput:5' 1035 '?'
DEBUG:__main__.Snapshot:    - new key
statusFlags: [0, 0, 0, 0]
DEBUG:__main__.Snapshot:upsert 6400 'analogInput:5' 'statusFlags' '[0, 0, 0, 0]'
DEBUG:__main__.Snapshot:    - new key
1037: ?
DEBUG:__main__.Snapshot:upsert 6400 'analogInput:5' 1037 '?'
DEBUG:__main__.Snapshot:    - new key
units: amperes
```

JoelBender, 2019-03-09:
Yep, those property identifiers are in the vendor range, so they stayed as integers rather than being auto-magically turned into enumeration names.  Try this version.

quest-gmulcahy, 2019-03-15:
Thanks, that seemed to resolve the sorting.

quest-gmulcahy, 2019-03-18:
I'm actually not sure if this should have been closed out. You might want to keep it open until all changes are merged.

JoelBender, 2019-11-17:
Version **0.17.7** has been released and everything has been merged in.  @quest-gmulcahy if you get a chance, please upgrade and test this again and if it works, close the issue.

quest-gmulcahy, 2020-02-21:
Sry, didn't realize I had to be the one to close the issue.

closed 2020-02-21 by quest-gmulcahy